alf45tar/PedalinoMini #154: Could not find remote branch feat/2.0.0 to clone

WernAir77, 2020-11-28:
Hey, 

first of all, thank you very much for this awesome project.

But I have an issue during the PlatformIO. build. Warning text:"Could not find remote branch feat/2.0.0 to clone" Do I need an extension? Or what I'm doing wrong?
It's my first time with Visual Studio.

alf45tar, 2020-11-29:
The latest branch is `master` which include PedalinoMini version 2.

WernAir77, 2020-11-29:
How can I change it? Pedalino.Uni?

WernAir77, 2020-11-29:
Now I have already 26 problems with unused variable 'checkPoint', 'channelNr.', ...

alf45tar, 2020-11-29:
Sorry I am lost. What version of PedalinoMini are you compiling? `master` branch or release `v1.0.0`?

Use `master` branch. Release `v1.0.0` is there only for historical reason.

closed 2020-11-29 by WernAir77