alf45tar/PedalinoMini #155: Error 1 during upload

WernAir77, 2020-11-29:
After solving some problems before, I'm actually not at the finish.

During upload Error 1 came. I my mind it is a connection Error between Mac and ESP32. (ESP32 DevKitC V2 from AZ Delivery).
The ESP is out of the box. Do I need to do some pre configuration, firmware updates, etc.? Or do I need to bring the ESP into the upload mode? When 'YES', how?

WernAir77, 2020-11-29:
esp32doit-devkit-v1-ble und esp32doit-devkit-v1 are uploaded. I installed some driver and after that, it worked.

So far so good, but how can I connect with Bluetooth or WiFi? iPhone and iPad can't find new hardware

alf45tar, 2020-11-29:
For Bluetooth connection you can use Garageband, MIDI Wrench or midimittr.
For Wifi connection setup you can use MIDI Network.

WernAir77, 2020-11-29:
All programs can't find the ESP. Actually I don't connect buttons, but in my mind it should be found as well, or am I wrong?

alf45tar, 2020-11-29:
esp32doit-devkit-v1-ble has no Wifi, only BLE.

esp32doit-devkit-v1 has Wifi and BLE. In order to use Wifi you must connect PedalinoMini to your router or connect your iPhone to PedalinoMini access point.

WernAir77, 2020-11-29:
How do I choose which esp32doit would be installed? 
It's my first time on Visual Studio and auch a project...

alf45tar, 2020-11-29:
[image]

WernAir77, 2020-11-29:
Thanks a lot. I installed the BLE, build and upload was successful, but I can't connect with the ESP32. 
The LED on the board is blinking, some bright, many short and low light. Is this normal, or is there an problem with the board?

alf45tar, 2020-11-29:
Did you have a display?
If not you can use the serial monitor.

WernAir77, 2020-11-29:
Did you mean "upload and monitor" or "monitor"?

alf45tar, 2020-11-30:
1. Re-sync PedalinoMini source code (platformio.ini has been updated yesterday).
2. Delete `libdeps`
3. Upload & Monitor

WernAir77, 2020-11-30:
Where do I find the libdeps? In the new downloaded PedalinoMini folder isn't such a file and no hidden files, etc...

alf45tar, 2020-11-30:
`libdeps` is where you build the PedalinoMini.

WernAir77, 2020-11-30:
Finally I found it. Command+Shift+. doesn't work at El Capitan 😬
Now it works fine and I can connect via bluetooth.

Thanks a lot

closed 2020-11-30 by alf45tar