alf45tar/PedalinoMini #157: Reboots repeatedly

someshout, 2020-12-02:
Hi alf45tar,
Thanks for your sharing this repository with us.

I followed your guide and have encountered a problem, My project is stuck in "Boot Mode" select stage and Esp32 reboots repeatedly.
I pressed the En & Boot button ,but it does not work.

Do you have any suggestions?

alf45tar, 2020-12-02:
It is solved here https://github.com/alf45tar/PedalinoMini/issues/155#issuecomment-735606131

someshout, 2020-12-02:
Thanks~ It's working!!
[image]
[image]

alf45tar, 2020-12-02:
I think you missed the last step Upload File System Image

https://github.com/alf45tar/PedalinoMini#how-to-build-and-upload

someshout, 2020-12-03:
Thanks~~~
[image]

closed 2020-12-03 by someshout