alf45tar/PedalinoMini #158: MIDI DIN not work

prostealth, 2020-12-02:
im just build pedalinomini usb and wifi works but MIDI out DIN port doesnt work.. does i need to make MIDI IN port too so it will work?

alf45tar, 2020-12-02:
No, you don't. MIDI IN is optional.
I suggest to check the wiring and check if the port is enabled in the WEB UI under Interfaces.

yush, 2020-12-09:
Hello,
I'm trying to use legacy MIDI out with no success too.
The wiring is OK, because I tried with a simple Arduino sketch. Moreover, I realised the optionnal part on the MIDI out schematics the LED is not blinking when a MIDI command is sent (I see it through the serial monitor). I also checked that the GPIO 4 is working with another simple Arduino sketch.

alf45tar, 2020-12-11:
I suppose MIDI OUT is enabled

[image]

Please attach a photo of your circuit.

alf45tar, 2021-01-06:
@yush did you solved?

yush, 2021-01-07:
Yes it works ! 
Sorry for the delay, the end of the year was quite busy. I tried again yesterday, and it works. I think my previous circuit was buggy.

Thank you very much for the support and the work you do on this project.

closed 2021-02-06 by alf45tar