alf45tar/PedalinoMini #159: I'm having issues with ESP Ble to upload

FabrizioGrasso0289, 2020-12-11:
I-m able to upload the first package using serial com4 but the esp Ble fails to uplaod.

additionally, the BLE package doesn-t have the upload image option than the Wifi option.

Attached you'll find the logs for build and upload

**Build**
>  Executing task: C:\Users\xxxxx\.platformio\penv\Scripts\pio.exe run --environment esp32doit-devkit-v1-ble <
> 
> Processing esp32doit-devkit-v1-ble (board: esp32doit-devkit-v1; platform: espressif32; framework: arduino)
> ---------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
> PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 3.10004.201016 (1.0.4)
>  - tool-esptoolpy 1.30000.201119 (3.0.0)
>  - toolchain-xtensa32 2.50200.80 (5.2.0)
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 44 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <AceButton> 1.8.0+sha.4a67456
> |-- <ArduinoJson> 6.16.0+sha.b85181a
> |-- <Battery Sense> 1.1.2+sha.b1b1492
> |-- <Blynk> 0.6.5+sha.73eb853
> |   |-- <WiFi> 1.0
> |   |-- <WiFiClientSecure> 1.0
> |   |   |-- <WiFi> 1.0
> |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
> |-- <ESP Async WebServer> 1.2.3+sha.f6fff3f
> |   |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
> |   |-- <FS> 1.0
> |   |-- <WiFi> 1.0
> |   |-- <ArduinoJson> 6.16.0+sha.b85181a
> |-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.1.0+sha.78bdcfb
> |   |-- <Wire> 1.0.1
> |   |-- <SPI> 1.0
> |-- <FastLED> 3.3.3+sha.fefc5f5
> |   |-- <SPI> 1.0
> |-- <MD_REncoder> 1.0.1+sha.d0c998d
> |-- <OSC> 0.0.0+sha.41b4498
> |-- <RemoteDebug> 3.0.5+sha.22f770f
> |   |-- <SPI> 1.0
> |   |-- <WiFi> 1.0
> |   |-- <WiFiClientSecure> 1.0
> |   |   |-- <WiFi> 1.0
> |-- <ResponsiveAnalogRead> 1.2.1+sha.504c16b
> |-- <NimBLE-Arduino> 1.0.2+sha.9074bba
> |-- <MIDI Library> 5.0.2+sha.d9149d1
> |-- <AppleMIDI> 2.2.0+sha.ee3e5ed
> |   |-- <MIDI Library> 5.0.2+sha.d9149d1
> |-- <BLE-MIDI> 2.1.0+sha.4aee670
> |   |-- <MIDI Library> 5.0.2+sha.d9149d1
> |   |-- <NimBLE-Arduino> 1.0.2+sha.9074bba
> |-- <ipMIDI> 1.0.3+sha.d131a37
> |   |-- <MIDI Library> 5.0.2+sha.d9149d1
> |-- <WiFi> 1.0
> |-- <Preferences> 1.0
> |-- <SPIFFS> 1.0
> |   |-- <FS> 1.0
> |-- <Ticker> 1.1
> |-- <ArduinoOTA> 1.0
> |   |-- <Update> 1.0
> |   |-- <WiFi> 1.0
> |   |-- <ESPmDNS> 1.0
> |   |   |-- <WiFi> 1.0
[... 104 lines cut ...]
> Serial port /dev/tty.SLAB_USBtoUART
> Traceback (most recent call last):
>   File "C:\Users\xxxxxx\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
>     _main()
>   File "C:\Users\xxxxxx\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
>     main()
>   File "C:\Users\xxxxxx\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
>     esp = chip_class(each_port, initial_baud, args.trace)
>   File "C:\Users\xxxxxx\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
>     self._port = serial.serial_for_url(port)
>   File "c:\users\xxxxxxx\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
>     instance.open()
>   File "c:\users\xxxxxxx\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
>     raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
> serial.serialutil.SerialException: could not open port '/dev/tty.SLAB_USBtoUART': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
> *** [upload] Error 1
> =========================================================== [FAILED] Took 13.04 seconds ===========================================================
> 
> Environment              Status    Duration
> -----------------------  --------  ------------
> esp32doit-devkit-v1-ble  FAILED    00:00:13.037
> ====================================================== 1 failed, 0 succeeded in 00:00:13.037 ======================================================
> The terminal process "C:\Users\xxxxxxx\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload', '--environment', 'esp32doit-devkit-v1-ble'" terminated with exit code: 1.
> 
> Terminal will be reused by tasks, press any key to close it.

alf45tar, 2020-12-11:
Change the upload port in platform.ini.
Are you on Windows or macOS?
The default port (/dev/tty.SLAB_USBtoUART) works for me on macOS. On Windows replace with COM4.

Cheers
alf45tar

FabrizioGrasso0289, 2020-12-11:
Im using Windows 10

alf45tar, 2020-12-11:
Your display is SSH1106

Comment this line
https://github.com/alf45tar/PedalinoMini/blob/b7cb0adb89cf4ed563ffa700f551c014ec91b316/platformio.ini#L11

Uncomment the next one
https://github.com/alf45tar/PedalinoMini/blob/b7cb0adb89cf4ed563ffa700f551c014ec91b316/platformio.ini#L12

FabrizioGrasso0289, 2020-12-11:
It worked. Do i need to have all the buttons already in the protoboard in order to start the esp in ble mode? 

As it is  now, I press and hold the boot button and release it in ble or normal mode but I doesn’t appear in my BLE-midi app, or midi wrench. 

I’m afraid I don’t see the device ID also to try the local host or access point. I don’t know how to configure esptouch in iPhone as well 

Sorry for all the trouble, you’ve been great help to go forward with this project

alf45tar, 2020-12-11:
I suppose you uploaded `esp32doit-devkit-v1` in order to have Wifi and BLE.
Buttons are not requested to connect to Wifi or using BLE.

I suggest to start with a factory reset to default (press BOOT for 15 seconds during restart). Then restart in normal mode (do not press BOOT).

You will see the wifi logo, the bluetooth logo, the SmartConfig countdown and the WPS countdown. After that sequence A.01 is on display.

In this situation BLE is ready to connect with midi wrench.

Please confirm and if not attach the boot log using Monitor.

FabrizioGrasso0289, 2020-12-11:
i used Clean All then Erase flash and build, upload and upload system image for devkit-v1 to ensure i had the proper version. Three tasks ended in success in about 40 seconds each. 

I see the wifi logo, then bluetooth and then it turns off using BOOT hold until the menu gets to factory reset
If i push EN button it does the above. 
---------------------------------
Monitor shows this repeteadly
 
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:8896
> load:0x40080400,len:5816
> entry 0x400806ac
> 
> Brownout detector was triggered

I'm using this ESP board
[https://www.aliexpress.com/item/32807887667.html?spm=a2g0s.9042311.0.0.293e4c4dSx7PNQ](url)

alf45tar, 2020-12-11:
`Brownout detector was triggered` is a clear sign of not enough power. Check your USB cable or USB port.
Try with a power supply or power bank.

FabrizioGrasso0289, 2020-12-12:
It worked with a 5v 2500mah power Supply. 

Now I see it in midi wrench as 9912CFA4. I was able to access through the AP and after I configured my wifi network in the options page I was able to access using the ‘deviceName.local’ method

I’m sending signals from midi wrench to pedalino and I see the changes on the screen

[image]

I don’t have the buttons connected yet to test sending midi signals to Bias Fx and I don’t thing I’ll be able to test it until next week as I’m waiting for the buttons+resistors to arrive

I cannot thank you enough for all the help you’ve given me! 

Looking forward to test all the possibilities to control bias fx on iOS and on PC to solder and enclose it for good!!

alf45tar, 2020-12-12:
Looking forward to see your complete build.

closed 2020-12-12 by alf45tar

Seaway10, 2020-12-15:
> Change the upload port in platform.ini.
> Are you on Windows or macOS?
> The default port (/dev/tty.SLAB_USBtoUART) works for me on macOS. On Windows replace with COM4.
> 
> Cheers
> alf45tar

Yes I am using windows 10 and I have changed in platformio.ini (/dev/tty.SLAB_USBtoUART) to COM4 but the same error:  name 'serial' is not defined , maybe somebody have any other solved for that trouble ?

denningja, 2021-03-02:
I had this problem, but solved it by opening Device Manager in Windows 10 settings. I found the USBtoUART was flagged with an error. I clicked on it and updated the driver (using Windows Update). It then popped up under Ports as COM3, so I replaced /dev/tty.SLAB_USBtoUART in its entirety with COM3. Then it worked!