alf45tar/PedalinoMini #160: TTGO T-Display - Cannot upload from PlatformIO

SD456, 2020-12-15:
Hello everyone,

I have just bought a TTGO T-Display board, and I have connected it to my PC, but I cannot upload the build with PlatformIO.
You can find the output of the terminal below.

```
> Executing task: C:\Users\dan\.platformio\penv\Scripts\pio.exe run --target uploadfs --environment ttgo-t-eight <

Processing ttgo-t-eight (platform: espressif32; framework: arduino; board: esp32dev)
-----------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AceButton> 1.8.0+sha.4a67456
|-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <Battery Sense> 1.1.2+sha.b1b1492
|-- <Blynk> 0.6.5+sha.73eb853
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <ESP Async WebServer> 1.2.3+sha.f6fff3f
|   |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.1.0+sha.78bdcfb
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <FastLED> 3.3.3+sha.fefc5f5
|   |-- <SPI> 1.0
|-- <MD_REncoder> 1.0.1+sha.d0c998d
|-- <OSC> 0.0.0+sha.41b4498
|-- <RemoteDebug> 3.0.5+sha.22f770f
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ResponsiveAnalogRead> 1.2.1+sha.504c16b
|-- <NimBLE-Arduino> 1.0.2+sha.9074bba
|-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <AppleMIDI> 2.2.0+sha.ee3e5ed
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <BLE-MIDI> 2.1.0+sha.4aee670
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|   |-- <NimBLE-Arduino> 1.0.2+sha.9074bba
|-- <ipMIDI> 1.0.3+sha.d131a37
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Ticker> 1.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <FS> 1.0
|-- <Update> 1.0
Building in release mode
Building SPIFFS image from 'data' directory to .pio\build\ttgo-t-eight\spiffs.bin
/blueboard.cfg
/css/bootstrap.min.css.gz
/favicon.ico.gz
/js/bootstrap.min.js.gz
/js/jquery-3.5.1.slim.min.js.gz
/js/popper.min.js.gz
/logo.png
/mustang.cfg
Looking for upload port...
Use manually specified: /dev/tty.SLAB_USBtoUART
Uploading .pio\build\ttgo-t-eight\spiffs.bin
esptool.py v3.0
Serial port /dev/tty.SLAB_USBtoUART
Traceback (most recent call last):
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
    _main()
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
    main()
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\dan\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "c:\users\dan\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/tty.SLAB_USBtoUART': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
*** [uploadfs] Error 1
============================================= [FAILED] Took 3.86 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
ttgo-t-eight   FAILED    00:00:03.861
======================================== 1 failed, 0 succeeded in 00:00:03.861 ========================================The terminal process "C:\Users\dan\.platformio\penv\Scripts\pio.exe 'run', '--target', 'uploadfs', '--environment', 'ttgo-t-eight'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Thank you for your advice in advance!

Kind regards,
Dan

alf45tar, 2020-12-15:
TTGO T-display is not supported by PedalinoMini. OLED is not SSD1309 or SSH1106.

Regarding the upload error check the following closed issue https://github.com/alf45tar/PedalinoMini/issues/159#issuecomment-743023952

SD456, 2020-12-15:
> TTGO T-display is not supported by PedalinoMini. OLED is not SSD1309 or SSH1106.
> 
> Regarding the upload error check the following closed issue [#159 (comment)](https://github.com/alf45tar/PedalinoMini/issues/159#issuecomment-743023952)

Thanks for the reply. I have tried to fix the upload error, but sadly it is still happening on COM4 too.

```
Retrieving maximum program size .pio\build\ttgo-t-eight\firmware.elf
Building .pio\build\ttgo-t-eight\firmware.bin
Checking size .pio\build\ttgo-t-eight\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.2% (used 63004 bytes from 327680 bytes)
Flash: [====      ]  37.4% (used 1420790 bytes from 3801088 bytes)
esptool.py v3.0
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM4
Uploading .pio\build\ttgo-t-eight\firmware.bin
esptool.py v3.0
Serial port COM4
Traceback (most recent call last):
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
    _main()
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
    main()
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\dan\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\dan\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "c:\users\dan\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
*** [upload] Error 1
============================================= [FAILED] Took 77.04 seconds =============================================
Environment    Status    Duration
-------------  --------  ------------
ttgo-t-eight   FAILED    00:01:17.041
======================================== 1 failed, 0 succeeded in 00:01:17.041 ========================================The terminal process "C:\Users\dan\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload', '--environment', 'ttgo-t-eight'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Kind regards,
Dan

alf45tar, 2020-12-15:
The COM number is computer dependant. It could be COM1, COM2, COM3, .... COM10 ... COM20 ...

Seaway10, 2020-12-26:
> The COM number is computer dependant. It could be COM1, COM2, COM3, .... COM10 ... COM20 ...
> 
> Check Devices in PIO Home

In device manager on to my pc COM4 is CP210x USBto UART even I trying to upload example project arduino-blink , but the problem is the same: 

Building in release mode
Building .pio\build\esp32dev\firmware.bin
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.7% (used 15348 bytes from 327680 bytes)
Flash: [==        ]  16.1% (used 210929 bytes from 1310720 bytes)
Traceback (most recent call last):
  File "C:\Users\vikto\.platformio\packages\tool-esptoolpy\esptool.py", line 39, in <module>
    if "serialization" in serial.__doc__ and "deserialization" in serial.__doc__:
NameError: name 'serial' is not defined
*** [.pio\build\esp32dev\firmware.bin] Error 1
================================================================== [FAILED] Took 4.77 seconds ==================================================================
The terminal process "C:\Users\vikto\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Seaway10, 2020-12-27:
> > The COM number is computer dependant. It could be COM1, COM2, COM3, .... COM10 ... COM20 ...
> > Check Devices in PIO Home
> 
> In device manager on to my pc COM4 is CP210x USBto UART even I trying to upload example project arduino-blink , but the problem is the same:
> 
> Building in release mode
> Building .pio\build\esp32dev\firmware.bin
> Retrieving maximum program size .pio\build\esp32dev\firmware.elf
> Checking size .pio\build\esp32dev\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM: [ ] 4.7% (used 15348 bytes from 327680 bytes)
> Flash: [== ] 16.1% (used 210929 bytes from 1310720 bytes)
> Traceback (most recent call last):
> File "C:\Users\vikto.platformio\packages\tool-esptoolpy\esptool.py", line 39, in
> if "serialization" in serial.**doc** and "deserialization" in serial.**doc**:
> NameError: name 'serial' is not defined
> *** [.pio\build\esp32dev\firmware.bin] Error 1
> ================================================================== [FAILED] Took 4.77 seconds ==================================================================
> The terminal process "C:\Users\vikto.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
> 
> Terminal will be reused by tasks, press any key to close it.

Ok the problem hase been resolved so if somebody will have tha same problem you need ;
Remove the folder C:\Users\(you profile in windows)\.platformio\packages\tool-esptoolpy\ fully and re-upload.
If that does not help, open a PlatformIO CLI and execute pio upgrade --dev and pio platform update. Close and re-open VSCode. Try to reupload the project.

alf45tar, 2020-12-27:
Good to know.

closed 2020-12-27 by alf45tar

denningja, 2021-03-02:
I have the same problem with a ttgo-t-display. I've deleted the tool-esptoolpy folder and tried uploading again; I've then used both those pio CLI commands to update, closed and re-opened VSCode and tried to reupload. I've also made sure the board is directly connected to my machine's USB port, not via a hub, just in case.

Here's the output:
`> Executing task: C:\Users\denni\.platformio\penv\Scripts\pio.exe run --target upload --environment ttgo-t-display <

Processing ttgo-t-display (platform: espressif32; framework: arduino; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10005.210223 (1.0.5)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AceButton> 1.8.2+sha.687e287   
|-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <ESP Async WebServer> 1.2.3+sha.2f78426
|   |-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.16.0+sha.b85181a
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.2.0+sha.7b73fde
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <FastLED> 3.4.0+sha.1fb0d7b
|   |-- <SPI> 1.0
|-- <MD_REncoder> 1.0.1+sha.d0c998d
|-- <NimBLE-Arduino> 1.2.0+sha.d18018f
|-- <OSC> 0.0.0+sha.41b4498
|-- <ResponsiveAnalogRead> 1.2.1+sha.504c16b
|-- <TFT_eSPI> 2.3.60+sha.fef40d9
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <AppleMIDI> 3.1.0+sha.ae9daba
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <BLE-MIDI> 2.1.0+sha.0cc25e9
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|   |-- <NimBLE-Arduino> 1.2.0+sha.d18018f
|-- <ipMIDI> 1.0.3+sha.d131a37
|   |-- <MIDI Library> 5.0.2+sha.d9149d1
|-- <Preferences> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Ticker> 1.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <FS> 1.0
|-- <Update> 1.0
Building in release mode
Retrieving maximum program size .pio\build\ttgo-t-display\firmware.elf
Checking size .pio\build\ttgo-t-display\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.2% (used 63076 bytes from 327680 bytes)
Flash: [========= ]  88.7% (used 1685845 bytes from 1900544 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/tty.SLAB_USBtoUART
Uploading .pio\build\ttgo-t-display\firmware.bin
esptool.py v3.0
Serial port /dev/tty.SLAB_USBtoUART
Traceback (most recent call last):
  File "C:\Users\denni\.platformio\packages\tool-esptoolpy\esptool.py", line 3969, in <module>
    _main()
  File "C:\Users\denni\.platformio\packages\tool-esptoolpy\esptool.py", line 3962, in _main
    main()
  File "C:\Users\denni\.platformio\packages\tool-esptoolpy\esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\denni\.platformio\packages\tool-esptoolpy\esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\users\denni\.platformio\penv\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "c:\users\denni\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/tty.SLAB_USBtoUART': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
*** [upload] Error 1
=================================================================================== [FAILED] Took 16.54 seconds ===================================================================================

Environment     Status    Duration
--------------  --------  ------------
ttgo-t-display  FAILED    00:00:16.542
============================================================================== 1 failed, 0 succeeded in 00:00:16.542 ==============================================`

alf45tar, 2021-03-03:
@denningja are you on a Mac?
If yes install the driver as indicated in platform.ini
If no replace /dev/tty.SLAB_USBtoUART with COMx in platform.ini.

denningja, 2021-03-03:
Thanks for the help. I'm on Windows - the problem was that it wasn't opening a port to the device because of a USB driver problem. I opened device manager and updated the driver for the USB device USBtoUART, which then caused it to pop up under Ports in device manager - which then showed the COM port number. I then did as you suggested and entered the COM port number in platform.ini. All working now, and I'm delighted to see all it can do! Hoping to use it to make a breath controller, amongst other things.