alf45tar/PedalinoMini #161: Pedalino Mini troubleshoot installing

Seaway10, 2020-12-15:
Hi there ! Could you some body to help me to upload project Pedalino Mini to my esp32. The point is I do everything like in manual but in beginning of installation i get an error . Python was install and Pyserial as well, but:
Traceback (most recent call last):
  File "C:\Users\vikto\.platformio\packages\tool-esptoolpy\esptool.py", line 39, in <module>
    if "serialization" in serial.__doc__ and "deserialization" in serial.__doc__:
NameError: name 'serial' is not defined
*** [.pio\build\esp32doit-devkit-v1-wifi\firmware.bin] Error 1
============================================================ [FAILED] Took 14.19 seconds ============================================================

Environment               Status    Duration
------------------------  --------  ------------
esp32doit-devkit-v1-wifi  FAILED    00:00:14.189
======================================================= 1 failed, 0 succeeded in 00:00:14.189 ======================================================= 
The terminal process "C:\Users\vikto\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload', '--environment', 'esp32doit-devkit-v1-wifi'" terminated with exit code: 1.

alf45tar, 2020-12-15:
Check the following closed issue https://github.com/alf45tar/PedalinoMini/issues/159#issuecomment-743023952

Seaway10, 2020-12-16:
I am using windows 10 and I have changed in platformio.ini (/dev/tty.SLAB_USBtoUART) to COM4 but the same error: name 'serial' is not defined , maybe somebody have any other solved for that trouble ?

Seaway10, 2020-12-16:
> Check the following closed issue [#159 (comment)](https://github.com/alf45tar/PedalinoMini/issues/159#issuecomment-743023952)

I am using windows 10 and I have changed in platformio.ini (/dev/tty.SLAB_USBtoUART) to COM4 but the same error: name 'serial' is not defined , maybe somebody have any other solved for that trouble ?

alf45tar, 2020-12-17:
It is already replied here https://github.com/alf45tar/PedalinoMini/issues/160#issuecomment-745410806

closed 2021-03-01 by alf45tar