alf45tar/PedalinoMini #162: Ladder Config

BotsmanDead, 2020-12-21:
Good day. I assembled the pedal according to the "TC HELICON Switch-6" scheme. Can't tune the resistance ladder. When you click on the knokki in turn, all the values are the same. (level 1-6) How can I resolve this issue?
P.S. I ask for a petition for incorrect speech) I am writing through Google translator) I am from Russia.

BotsmanDead, 2020-12-21:
when measuring resistors, I noticed that they have an error. 8.7 kΩ 9.2 kΩ ...
Could this affect circuit performance?

alf45tar, 2020-12-21:
Try to switch ladder connection to digital and analog pin.
Resistors difference is too high for 10K resistors however calibration is designed for compensate this kind of difference.

alf45tar, 2021-01-06:
@BotsmanDead did you solved?

juani13973, 2021-02-03:
Dear Alf45tar, first of all, thank you very much for this peace of art you made !. I discovered it 2 weeks ago and here I am trying to make it work.
I think I have the same problem as BotsmanDead. I configured 3 leaders of 6 switches following the TC-Helicon diagram with 10K resistors.

I calibrate them using the option 7 and I think it worked. This is what I got in options :
[image]

This is how I configured the pedals (the Analog is working perfect) but the pedal page shows a 7th pedal.... and we should see 6 right ?:
[image]
[image]

This is the screenshot of Pedal 1 with 6 switches sending a CC for each one....
When I press any of the 6 switches of each leader it sends the first CC value with 0 or 127 alternatively.

This is how I built the ladder and I crossed the cables Tip and Ring as said in the diagram (and I tried one way and the other and it didn't work):
[image]
[image]

I am not an expert and I think I uploaded the right version:
[image]

I do not know how to send you more data... Sorry.
One other question, not related to this topic, since I will have 18 switches, can I have one led associated with each one ?

quick demo of the what is going on:

https://user-images.githubusercontent.com/77895272/106813779-c5c08c00-6671-11eb-94fe-2c2eba95b2ec.mp4

Again, 1000 thanks for this massive work !!!
Juan,

BotsmanDead, 2021-02-04:
> @BotsmanDead did you solved?

I haven't been able to solve this problem.

alf45tar, 2021-02-04:
The 7th pedal is the BOOT Button on board.
The numbers of leds is definend in Pedalino.h

juani13973, 2021-02-04:
Thx alf45tar !
Could you comment about the ladder not working with 6 switches  ?
Again, thanks a lot !

mknerr, 2021-02-04:
I think something is￼ wrong with your￼ call ladder switch. The values you showed in the picture ￼should all be different.￼
I'll admit, this isn't my strong suit, but looking at your Proto type, it seems like you have continuity throughout all the resistors regardless of your switches.￼

alf45tar, 2021-02-05:
Ladder looks good.
Can you confirm that GREEN cable is connected to PIN_A1 (GPIO 36) and ORANGE cable is connected to PIN_D1 (GPIO 25)?

juani13973, 2021-02-06:
Hi Alf45tar, thanks for replying back. I confirmed it is the case.
I think that when I booted in mode 7 to config the ladder, the resistors were correctly picked as per picture above.
I used the scheme of 6 switches with all 10K resistors. Should I try Voltage Ladder 5 with all different resistors and add one more switch ? (I bought resistors and they are arriving tomorrow, I would try but i do not think that would the be issue)

Thanks again.

juani13973, 2021-02-08:
Hi to all, i managed to solve the problem -> I was using the option 7 to configure the ladder in the wrong way : *YOU NEED TO WAIT PRESSING UNTIL THE TIME PASS AND THE LADDER VALUE IS REGISTERED IN EACH SWITCH*.
I tried with both examples, the 10K 6 switches ladder TC-Helicon and the other example with 5 switches.
I think the other example is better. The switches would be ordered left to right and the distance between registered values is higher which It could provide less room for noise... (i do not know....)

Thanks !! I move to the leds now :)
Quick video that shows how to configure the ladder.

https://user-images.githubusercontent.com/77895272/107289045-bae86b80-6a64-11eb-8ca5-0196f011eca7.mp4

alf45tar, 2021-02-09:
You are right. There is lacking of documentation.
Thanks @juani13973 for the explanatory video too.

The 10K resistor ladder is a replica of TC-Helicon pedal. I do not suggest it too.
Any ladder pedal (including custom and DIY) can be used using ladder configuration.

closed 2021-03-01 by alf45tar