alf45tar/PedalinoMini #164: Strange Acton WI behavier

MaxZak2020, 2020-12-27:
Dear Alf45tar! 
Thanks for you great work with tis progect. I've made it and it works generaly. But I have some issues during tunning it to my goals.

Whaen I tryed to configure ladder with 6 buttons first time "Apply" and "Save" disappiar when I added Button 6. [Pic 1](https://yadi.sk/i/XAfCVsVQfi_ycA)

But when I've deleted Pedal 2 those buttons are appeared again and I could save it but when I added Pedal 2,  "Apply" and "Save" disappiared again. [Pic 2](https://yadi.sk/i/anjdub3PPdPcRw)

But after I've deleted and added Pedal 2 it again looks like suppose to be... [Pic 3](https://yadi.sk/i/bU3TMYb0yhsTvg) 

It's strange and can to confuse))

alf45tar, 2020-12-27:
Thanks @MaxZak2020 for your detailed contribution. I will have a look.

alf45tar, 2020-12-28:
A long list of actions requires a lot of memory to create the web page. Sometime the memory is not enough to complete the page creation and the page is incomplete.
To avoid it I suggest:
1. Use the filter at the top to display one pedal per time
2. Boot in WIFi mode to free Bluetooth memory during configuration

I am exploring alternatives like creating page in trunks (I am already sending page in trunk).

Regards
alf45tar

MaxZak2020, 2020-12-28:
I see! Ok Thanks!

closed 2020-12-28 by MaxZak2020