alf45tar/PedalinoMini #165: BiasFX LEDs management

MaxZak2020, 2020-12-27:
Dear alf45tar!
Thank you for your wonderful work with this project. I've done this and it works overall. My goal is to switch presets in Bias FX Mobile IOS. And the Pedalino do it well. But there is one point with LEDs that needs to be discussed. 
When you switch another button, it switches presets in Bias FX, but the led of the previous button keeps the same state (on or off). Is it possible to add a configuration for the LEDs when they will turn off when another button is pressed? 
Or, if this option is already implemented, could you explain please how to configure it. 

Thank you in advance!

MaxZak2020, 2020-12-27:
It could be just option “Exclusive status” in button configuration dialog - “On” (turn off other leds) or “off“ (leave other leds us is).

alf45tar, 2020-12-28:
PedalinoMini is reacting on MIDI events. If it receives a MIDI message associated with a pedal is switch its status on or off.
Unlikely Bias FX is not sending any MIDI out.

The solutions could be to add pedal relationships.

MaxZak2020, 2020-12-29:
Have an idea - for every 6 buttons make additional 5 "Releases" for other LEDs with color "Black". 6 * 5= 30 additional Rows in Action List. I wonder would it be enough memory for that trick. Anyway - not so beautiful solution of the problem ://

alf45tar, 2021-01-01:
I forgotten there is another option (https://github.com/alf45tar/PedalinoMini/issues/104#issuecomment-673485126): use Program Change. A PC message will off all the others leds associated to a PC message too.

MaxZak2020, 2021-01-02:
OK, thanks. It may be usefull but not for BiasFX - this soft can recieve only CC. But it's possible to use midi-mapping software I suppose.

closed 2021-01-04 by MaxZak2020