alf45tar/PedalinoMini #166: Example Bias fx2 and Pedalinomini

Seaway10, 2020-12-27:
Finally I uploaded Pedalino mini to my esp 32 but unfortunately can't find manual how to connect it to Bias Fx2 for example , sorry for stupid question but i do that at first time ) could somebody help me , and many thanks in advance !

MaxZak2020, 2020-12-27:
If you try to connect with BiasFX iOS mobile you can do it via Bluetooth LE but you need to use Midimittr app from AppStore along with Bias to transmit CC from BLE  of Peddalino to Bias. It’s free.

Seaway10, 2020-12-27:
> Если вы попытаетесь подключиться к мобильному устройству BiasFX iOS, вы можете сделать это через Bluetooth LE, но вам нужно использовать приложение Midimittr из AppStore вместе с Bias для передачи CC из BLE Peddalino в Bias. Это бесплатно.

Many thanks for your reply , but I am using looptop and Windows 10 (

alf45tar, 2020-12-27:
Use https://www.tobias-erichsen.de/software/rtpmidi.html for rtpMIDI/AppleMIDI or https://xmmc.de/mnet/ for ipMIDI

closed 2021-03-01 by alf45tar