alf45tar/PedalinoMini #169: Wrong serial midi data via USB

davethegrinch, 2020-12-31:
Hello, 
Firstly, thanks for all the great work you've put into this. 

The problem I'm seeing is that sending midi data via serial USB is not working correctly. I have a Sparkfun ESP32 Thing and a Sparkfun Micro Pro configured as the instructions say. 

When I operate an analog pot, I see the right midi CC values generated on the ESP32 but when I monitor in the incoming midi on my mac that CC messages are replaced by a stream of other midi messages such as Program Change, note off, CC on a random selection of channels. 

[image]

If I turn on IPMidi interface and monitor that from my mac, I see exactly the right messages. 
[image]

Something is not translating well from Micro Pro board. I have the UsbMidiConvertor library loaded and I can program that to send hardcoded midi message just fine. I've matched the baud rate in the convertor to ESPSerialMIDI.h (115200). But something between the micro pro and ESP32 board is not working even though they share the same GND. 

Any ideas??

Thanks...
Dave

alf45tar, 2020-12-31:
Are ESP32 board and Pro Micro sharing the GND?

davethegrinch, 2020-12-31:
I believe so. I have them both connected to a common GND bus.

davethegrinch, 2020-12-31:
I think I might have a ground loop somewhere.

davethegrinch, 2021-02-09:
Well - turns out it was a bad pro micro board. Issue closed

closed 2021-02-09 by davethegrinch