alf45tar/PedalinoMini #170: Legacy MIDI-out latency

MaxZak2020, 2021-01-05:
I have sad experience with looper in Ableton Live, wich I trigger with PedalinoMINI from MIDI out -> MIDI USB adapter. The latency is tremendows - about 1sec or more. It's imposible to use. How can it be and what to do? 
Thanks in advance for any ideas.

_Originally posted by @MaxZak2020 in https://github.com/alf45tar/PedalinoMini/issues/89#issuecomment-754204189_

alf45tar, 2021-01-05:
If USB midi works it is strange to have so much delay. I suspect a different reason.
Check the pedal configuration. For example if double press is enable, Pedalino is waiting Double Press Time before release the single press event (Pedalino is waiting the second press that will never happen).
Double Press reduce the responsiveness. However the timing is configurable according your need: very fast clicking or relaxed.

MaxZak2020, 2021-01-05:
I've configured Long press for this pedal and all buttons of it. Maybe sittuation is the same as for Double press. So I try to fix...

MaxZak2020, 2021-01-05:
Yes! This is it. Alfastar, thank you for the support and project again!

closed 2021-01-05 by MaxZak2020