alf45tar/PedalinoMini #171: How to switch profiles?

MaxZak2020, 2021-01-06:
Now Boot button switch Banks as I can see. But is there still way to switch profiles without Web UI? 
I need to swich it quickly because of different pedal config for BiasFX and Looper operations.

alf45tar, 2021-01-06:
No way for the moment if you don't use a TTGO T-Eight. I am working to add profile switching as global pedal function.

alf45tar, 2021-01-06:
if you want give a try I just released an important change. Onboard buttons are fully configurable as pedal 7, 8, ...
BOOT button is now pedal 7 and can be assigned to any function including Profile+ or Profile-.

MaxZak2020, 2021-01-06:
Yes, sure. BTW I've found a way to use Not On and Off events to operate looper. And it doesn't matter how  Pedals are configured (even if Long Press or Double Press are on), so now I can use only Banks ))

MaxZak2020, 2021-01-07:
I've refreshed repository but got to errors. Should I ignore them or they could be fatal for normal condition of Pedalino ufter re-uploading? (Sorry for missunderstanding. I'm absolute beginer in codding)

[image]
[image]

alf45tar, 2021-01-07:
You can ignore them. They are non blocking errors because 'include' subfolder is missing in PedalinoMini folder.

MaxZak2020, 2021-01-07:
And would you please point out what strings to change to reasign button BOOT? Earlier it was from GPIO_NUM_0 to GPIO_NUM_13 (D6)

alf45tar, 2021-01-07:
You don't need anymore to change the code.
If you to use D6 to switch Profile goto Pedals page and on Pedal 6 set Single Press to Profile+.

MaxZak2020, 2021-01-07:
I just want use booting mode sequence (EN -> BOOT hold) and to escape using buttons on the ESP32 board inside the box I brought two buttons on front pannel.

alf45tar, 2021-01-07:
The booting sequence is activated using the FACTORY_DEFAULT_PIN

MaxZak2020, 2021-01-07:
Sorry, I think I've burned out microUSB port on the board, so now I'm not able to reload anyting to board. 
I've got short circuit of power with smoke and burning of something on bord %)))) (didn't search what it is yet) couple days before and now microUSB is dead as I can see. There is no device apiares after connecting board to USB (it was on com3 before). 
Nevertheless bord is working good.
Need to order new anyway.

PS New build was successful and I've tryed to load firmware.bin after build via HTTP but "Update fail!" :/

closed 2021-02-27 by MaxZak2020