alf45tar/PedalinoMini #173: Bank Select +/-

MaxZak2020, 2021-01-25:
Hi there. Unfortunately I can not understand how to use Bank Select + to change bank of Program patches in midi unit. Would you please explain how shoul I use MSB and LSB in both casese (+ and -)

I've tryed like that: MSB=0 and LSB = 0-127 for + and -
But Tone Stack (guitar processor iOS soft) blinks to next Bank and then back to Bank 0 when push + (for - it is the same but negative - it lows down the bank and then switches to 0).

alf45tar, 2021-01-25:
@MaxZak2020 try the latest version.
It should work as documented.

MaxZak2020, 2021-01-27:
Thanks a lot. I will try after new ESP32 board has come.. Old one has usb port burned.

alf45tar, 2021-02-27:
@MaxZak2020 may I close the issue?

closed 2021-02-27 by MaxZak2020

bobvc133, 2023-02-18:
> @MaxZak2020 try the latest version. It should work as documented.
BOSS GT1000CORE cannot use the BNAK SELECT+/- function normally, is this function working normally now?

[image]
[image]
[image]

https://www.thegearpage.net/board/index.php?threads/boss-gt-1000core-midi-mapping-banks-1-4-do-they-actually-work-solved.2375020/