alf45tar/PedalinoMini #175: Display only shows first button of ladder

joesuspense, 2021-01-26:
I have my ladder setup and working great, but the display only shows the first button in the ladder. For instance, the action tag for button 1 shows on the display, or the current value of button 1, but none of the others show up. I loaded the "mustang" profile and noticed that all the single pedals show up on the display. It only seems to occur for ladders.

[image]
[image]
[image]

mknerr, 2021-01-27:
It's my understanding that those tags only correspond to the first button for any given input, although things may have changed since I was more actively experimenting.

￼It would be nice to be able to do that, but if you think about it, if you have other switches attached, where would their current state display? It's a bit a "catch 22".

joesuspense, 2021-01-27:
This is true. I thought about how it could be done differently after posting and couldn’t think of a better option. Maybe  label it “pedal 1” or ladder?

mknerr, 2021-01-27:
I went through this too, and can't say I have it all figured out, but I think one needs to get away from trying to group functions based on the input. Especially, since now all of the functions can be set independently. ￼
You could try and set it up so the first pedal on each input provides the the functions where you would really want that visual feedback. 

Then, if possible, use the "extra" ones where it is not as crucial, or perhaps where you can get that feedback from the app.

joesuspense, 2021-01-27:
In my setup, I only have a 5 button ladder attached to one input. I can totally live with this, now that I understand how it's working. Thanks!

closed 2021-01-27 by joesuspense