alf45tar/PedalinoMini #176: Legacy MIDI need 2 clicks

jiosta, 2021-01-29:
Hi,
Thank you for your hard work. It is a very nice project.
I have the following issue:
I configured a Pedal to send CC Message with value 1 on channel 1 to control my BOSS GT1000Core.
I configured From Value 0 to Value 127 and Tag When Off 127 and Tag When On 0.
This means, that it should send alternating 0 127 whenever i click.
This works, but I always have to click twice, for the boss to receive the message.
Is this a problem on my boss device or my pedalino?
I´m using esp32doit-devkit-v1
This is he Terminal output of 4 clicks:
Pedal: 3     Button: 0    EventType: 0     ButtonState: 0
Pedal: 3     Button: 0    EventType: 1     ButtonState: 1
Pedal: 3     Button: 0    EventType: 2     ButtonState: 1
Action: 0
CONTROL CHANGE.....Code   1......Value   0.....Channel  1
Pedal: 3     Button: 0    EventType: 0     ButtonState: 0
Pedal: 3     Button: 0    EventType: 1     ButtonState: 1
Pedal: 3     Button: 0    EventType: 2     ButtonState: 1
Action: 0
CONTROL CHANGE.....Code   1......Value 127.....Channel  1
Pedal: 3     Button: 0    EventType: 0     ButtonState: 0
Pedal: 3     Button: 0    EventType: 1     ButtonState: 1
Pedal: 3     Button: 0    EventType: 2     ButtonState: 1
Action: 0
CONTROL CHANGE.....Code   1......Value   0.....Channel  1
Pedal: 3     Button: 0    EventType: 0     ButtonState: 0
Pedal: 3     Button: 0    EventType: 1     ButtonState: 1
Pedal: 3     Button: 0    EventType: 2     ButtonState: 1
Action: 0
CONTROL CHANGE.....Code   1......Value 127.....Channel  1
Here is how I configured the Pedal
[image]

alf45tar, 2021-01-30:
Looking the log PedalinoMini works as designed. I cannot support you with MIDI implementation of BOSS GT1000Core.
Just a note: Tag When Off/On is just a a label. It is used only on display. I don't know why you inverted the tag with the value.

jiosta, 2021-01-30:
Ok, thank you for your reply.
I also think, that pedalino sent everything properly.
Whenever I clicked, the LED connected to the Legacy MIDI Port was lighting up, so I´m quite sure, that the message was sent.
So I probably have configured something wrong on the receiving Pedal, or it wants another Message.
I´ll send an update, in case I can set it up properly

mknerr, 2021-01-30:
Maybe you can hook it up to a computer or other device running a MIDI monitor so you can see exactly what's going on.

jiosta, 2021-02-02:
It is kinda weird, but I managed to get it to work properly by configuring, that the same CC is sent when press and release are triggered.
Maybe Boss Pedals need 2 commands, I don´t know, but it works now.
Here´s the configurations, in case you are interested.

[image]

alf45tar, 2021-02-02:
It looks like that Boss pedals switch only on 127 (for some pedals a value greater than 63 is enough).
Put 127 in both From Value and To Value and one actions per pedal should be enough.

jiosta, 2021-02-02:
I´ll try as soon as I can, thank you.
I think you can close the issue

closed 2021-02-06 by alf45tar