alf45tar/PedalinoMini #181: Problem with legacy midi

teopost, 2021-02-06:
In the mail README I read:

_Legacy serial MIDI interface with 5 pins DIN connectors. Requires additional hardware (see schematic)._

[image]

but in the diagram it seems to me that it is possible to use the legacy midi without using other hardware.

I am wrong ?

I ask this because I can't get it to work!

[image]

alf45tar, 2021-02-06:
Additional hardware means a couple of resistors and a connector.
Di you tried to invert blank and red wire?

teopost, 2021-02-06:
No. it seems to me that it is correct

[image]

[image]

maybe the configuration is wrong?

Configuration

[image]

Pedals

[image]

Actions

[image]

alf45tar, 2021-02-06:
May I suggest to add the optional led to see the MIDI traffic?

teopost, 2021-02-06:
the led is always off

[image]

teopost, 2021-02-06:
@alf45tar the configuration is correct ?

alf45tar, 2021-02-06:
Yes, it is

alf45tar, 2021-02-06:
Yes, it is

teopost, 2021-02-06:
It has been compiled since the last commit

[image]

The oled display is a wemos oled shield that I had. I already know it's not supported.

I have to buy a compatible one but first I have to get the legacy midi to work

teopost, 2021-02-07:
erasing the eeprom it works!

https://www.youtube.com/watch?v=DIaNEkbmiro&feature=youtu.be

Thank you! Fantastic project!!!!

closed 2021-02-07 by teopost