alf45tar/PedalinoMini #182: Problem in configuring Sequences

teopost, 2021-02-07:
I have configured a pedal to perform a Sequence

[image]

[image]

it seems that the value sent to the midi is always 2, that is the one configured in Pedals and not in the Sequences

[image]

What am I doing wrong ?

alf45tar, 2021-02-08:
You are not doing anything wrong.
It is by design. Every CC in a sequence is transmitting the value coming from the pedal. It makes sense if it connected to a Analog or Jog Wheel pedal. It doesn't make sense for Momentary/Latch pedals.
It is an historical feature before the Actions era. With multiple Actions by pedal it is not anymore requested.
I will fix it.

teopost, 2021-02-08:
Can I only use actions to send a sequence of messages?

alf45tar, 2021-02-08:
Yes, you can if the numbers of messages to send it is not too long.
Otherwise a Sequence is better.

teopost, 2021-02-08:
if I execute 2 commands in sequence in this way using Actions, the device I am controlling does not accept them.

[image]

It only works if I run the second command with "On Release"

[image]

I think a short sleep from one command to another is needed

I did a test with another arduino project

[image]

I think it would be useful to be able to add a pause between CC commands

teopost, 2021-02-08:
Wow! Great buy!

davidin73, 2021-02-08:
I'm interested about this problem because my intention is to have more than 3 snapshots with the commands sequences,as rhemorningstar' products

teopost, 2021-02-08:
AHHH! I have the same problem too!

teopost, 2021-02-08:
HX stomp does not have a native patch up/dn midi command so we can switching to scroll mode and emulating a FS1/2 button press.
I did this with this project

* https://github.com/mattzzw/midi_muppet_hx_2btn

davidin73, 2021-02-08:
Hm, no....i wanna be have a different approach.
The concept is :if I can have different variations of the same preset with the snapshot,i can have 'more snapshots' by changing the various blocks parameters, with a sequence of CC..
And/or by changing a preset with the trails activated.
I think I can do this with the cc sequences, to avoid the silence between the programs/presets change

teopost, 2021-02-08:
PedalinoMini is better than morning star!

davidin73, 2021-02-08:
Sure! This is why I'm building a pedalino mini for my stomp🤗

teopost, 2021-02-08:
I'll show you my next pedalboard just finished

teopost, 2021-02-08:
> Sure! This is why I'm building a pedalino mini for my stomp🤗

How many switches?

closed 2021-02-08 by teopost

davidin73, 2021-02-09:
8 swirches