alf45tar/PedalinoMini #184: Problem configuring midi block bypass with HX Stomp

teopost, 2021-02-13:
Is possible configuring HX Stomp to bypass a block using midi commands.

From manual: "CC values 0-63 turn the block off; values 64-127 turn the block on."

If I configure a pedal with "From value" to 5 and "To value" to 65 the block turns off when the pedal is ON (and viceversa)

I inverted the values by putting "From value" to 65 and "To value" to 5 but when I do it pedalinomini doesn't work and the display goes crazy!

See video here:  https://www.youtube.com/watch?v=BB86vlJ_4RQ&feature=youtu.be

alf45tar, 2021-02-13:
Thanks for feedback. I am going to fix.

alf45tar, 2021-02-13:
It was only a display issue. Fixed.

teopost, 2021-02-13:
Tested!
It works. Thank you

closed 2021-02-13 by teopost