alf45tar/PedalinoMini #186: Request pedalinomini logo in .svg format

teopost, 2021-02-17:
Hi,
  my first pedalboard with PedalinoMini is almost finished !
I would like to thank you by putting the PedalinoMini logo 

[image]

Can you provide me the svg file?

alf45tar, 2021-02-17:
Just uploaded. Check under images folder.
I am excited to see the final assembly.

teopost, 2021-02-17:
I have some pictures on its construction. if you need them tell me where to put them

alf45tar, 2021-02-17:
Upload here or create a separate issue/discussion for your build.
I will add one picture on the home page.

teopost, 2021-02-19:
I still don't know how to put the logo!

[image]

teopost, 2021-02-26:
[image]

[image]

https://www.youtube.com/watch?v=XpELX5sSgZE&feature=youtu.be

davidin73, 2021-02-26:
Ho Stefano.
what kind of container did you use for pedalino? (excellent result by the
way ...)

teopost, 2021-02-26:
Hi,
   I uploaded all the images here 

https://photos.app.goo.gl/4xGTSbXWA1XYLDvw9

Enclosure is this

https://www.amazon.it/dp/B0827VQQ69?ref=ppx_pop_mob_ap_share

Ps: ma sei italiano ?

Inviato da iPhone

davidin73, 2021-02-26:
Di Trieste 😁😁😁

alf45tar, 2021-02-27:
@teopost 
Awesome job.

closed 2021-03-01 by teopost

jimhiggs, 2021-03-04:
Did you include wifi and bluetooth? and if so how well do they work inside the case?

teopost, 2021-03-04:
I use only wifi and work fine

jimhiggs, 2021-03-04:
Thanks. I made the little black one with red buttons and the Pedalinomini tm on the front. Have added a rotary encoder for bank select and usb midi and use a TC Helicon Switch 6 with it. Think I will build a new version with 8 switches and sell the switch 6. I will also make one of those 2 button switches in your link for my HX Stomp. Useful little pedal.

jimhiggs, 2021-03-04:
I will probably add midi thru as well so can use both at once.  It would be really something if I could put a bigger screen on it though, the info screens are too small for me to read so I just use patch/bank names. The full range of midi options on the padalinomini are awesome though. I have driven my Hx using rptmidi over wifi between the box and my pc and usbmidi to Hx from my pc. I have also used the Bluetooth midi. Fantastic.