alf45tar/PedalinoMini #188: TC ladder issue/challenge

wrpooley, 2021-02-24:
I have limited experience with reading circuits but I've basically duplicated the design of BotsmanDead for the Ladder6 resistor ladder.  However, when I boot into Ladder Config and calibrate each button, I only Pedal:0 and Button:0 for each of the 6 buttons.  Also, the Ladder Level values posted in the serial monitor on re-boot are oddly close together for the first 4 values (ascending) and the same for value 5 & 6 (see screenshot below).  

If anyone would be so kind as to review the attached media and see of there is anything wrong with my wiring?  One are of confusion for me is the +3V and 10K resistor show on an analog pin (A1, etc).  If it's not clear form the pictures, I have GND going though a 10k resistor to the start of the ladder, one leg of the switch going to D1 (GPIO_25) and the other leg of all the switches connected together in series and connected to A1 (GPIO_36).

I appreciate any help in this matter.  And if anyone has a good resource for someone trying to learn how to build a circuit from a schematic I'd appreciate that too!
[image]
[image]
[image]

https://user-images.githubusercontent.com/61659362/108926565-ae3f4800-760c-11eb-9660-f907cd025f5b.mp4

alf45tar, 2021-02-24:
Is the brown resistor of the ladder of 10k? It looks like is only 1k.

wrpooley, 2021-02-24:
Yes.  It measure 10k on multi-meter.  I could certainly swap out for same
type as ladder.

rigr, 2021-02-24:
Just a quick idea: Does the resistor in red circle go to the correct hole
in the breadboard?

rigr, 2021-02-24:
oh, attachment not visible - here it comes:
[image]

rigr, 2021-02-24:
well, just saw that in the video it looks like it is connected to the correct hole.

wrpooley, 2021-02-24:
Yes I believe it is.

alf45tar, 2021-02-24:
It must be connected to line 17

wrpooley, 2021-02-24:
That is correct. I also made all 10k the same type. No difference.

alf45tar, 2021-02-24:
I don't see the pull-up resistor for A1. It is a 10k resistor from A1 (GPIO_36) and +3.3V

wrpooley, 2021-02-24:
I’m afraid this is where my knowledge of circuit design breaks down!

Does that mean I simply put a 10k resistor between the 3V rail and A1 and
leave the connector from the end of the switches also to A1? If I do that,
then during the calibration, all of the switches ladder level is 1010

Grateful for you hand holding!

alf45tar, 2021-02-24:
Because the orange wire is not connected to GND.

wrpooley, 2021-02-24:
DOH!  That did it. Thank you so much.

closed 2021-03-01 by alf45tar