alf45tar/PedalinoMini #190: Expression Pedal

Seaway10, 2021-02-26:
Good day! Could someone explane detailed how to connect expression pedal to pedallino I have M-Audio pedal https://m-audio.com/products/view/ex-p I try to do that but it doesn't work I use schem  https: //raw.githubusercontent.com/alf45tar/PedalinoMini/c074031343907d6ac2c8218f83410afacf81d921/images/Schematic_PedalinoMini_Sheet-3.svg
as example to connect Expression PedalYamaha FC-7 and Expression PedalRoland EV-5 maybe somebody show in video or schematick pictures hot to do that .

alf45tar, 2021-02-26:
TIP to A1 
RING to D1
SLEEVE to GND

Seaway10, 2021-02-26:
As I understood correctly 
 TIP to A1
RING to D1
SLEEVE to GND
it's for Boss FS-5U this is foot swich but my expression pedal id Volume pedal and I Used before your reply next scheme:
TIP D6 
RIG A6 +3.3V across resistor 10kOm and
SLEEVE ground

alf45tar, 2021-02-26:
https://github.com/alf45tar/Pedalino#expression-pedals

Seaway10, 2021-02-26:
ok thanks a lot I will try to do that and one more do I need to do any presettings into admin pannel of pedallino for expression pedal ?

alf45tar, 2021-02-27:
Set Pedal 1 as Analog and enable Analog Calibration

Seaway10, 2021-02-27:
Ok looks like connected like you said but on button one the numbers of value start jumping and  my mother jack of the expression pedal i using is 
https://www.google.com/imgres?imgurl=https%3A%2F%2Fimg.muzline.ua%2Fimage%2Fcache%2Fcatalog%2FSKCC217-600x315.jpg&imgrefurl=https%3A%2F%2Fmuzline.ua%2Fsoundking-skcc217%2F&tbnid=zvTm1TeuT5lRFM&vet=12ahUKEwjvztPi_InvAhVByyoKHddiDIcQMygAegQIARAj..i&docid=QZf9-59sN8AVwM&w=600&h=315&itg=1&q=SoundKing%20SKCC217&hl=en&ved=2ahUKEwjvztPi_InvAhVByyoKHddiDIcQMygAegQIARAj

maybe it's wrong and do need to use stereo ?

Seaway10, 2021-03-01:
In manual have a some remark:
Do not forget the add the pull-up resistors on PIN_A1 to PIN_A6 otherwise pins will be floating. A floating pin can trigger unexpected MIDI events. As alternative you can disable the not used pedals via web interface.
What is a nominal of the pull-up resistor should be to exclude floating parrameters ? 10 kOm?

alf45tar, 2021-03-01:
For excluding floating any resistor greater than 220 ohm is enough. I suggest 10K ohm to reduce the power consumption when power by USB. 50K could be ok too.

closed 2021-04-01 by alf45tar