alf45tar/PedalinoMini #191: Web interface problem adding actions

teopost, 2021-02-27:
By adding a new action to a pedal, it is created with a blank value in the "On" and "Send" fields
By refreshing a new action is also added.
I attach video:

https://user-images.githubusercontent.com/2573389/109382979-e338ec80-78e3-11eb-95e8-2c1587364561.mov

alf45tar, 2021-02-27:
You are not using the latest version. Please try with the latest.

teopost, 2021-02-27:
Ok. I still have to figure out how to update the firmware without opening the pedal!
Thank you

alf45tar, 2021-02-27:
Pedalino support firmware update Over The Air.

1. Connect to http://pedalino.local/update
2. Locate the firmware.bin file
3. Upload

teopost, 2021-02-27:
I have some problems loading the firmware.
I try to open the box

[image]

teopost, 2021-02-27:
It doesn't work even with the enclosure open

alf45tar, 2021-02-27:
What is not working?
If it is open connect the usb cable to upload new firmware.

teopost, 2021-02-27:
What is the difference between "Upload filesystem image" and "Upload filesystem image OTA" ?

[image]

alf45tar, 2021-02-27:
To upload the firmare use Upload.
The filesystem image is the 'data' folder only to be upload separately.
OTA stand for Over The Air but you need to configure how to connect into platformio.ini

teopost, 2021-02-27:
I'm excited. Pedalinomini is fantastic!

teopost, 2021-02-27:
I don't understand why, but OTA upload doesn't work for me.
I also tried with firefox

Seaway10, 2021-02-27:
Is it possible to upload OTA file for update not use platformio ?

alf45tar, 2021-02-27:
Attach the serial log during update and I will have a look.

teopost, 2021-02-27:
```
> Executing task: pio device monitor --environment esp32doit-devkit-v1 <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
���␖��␚�len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ChipRevision 0, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 252544, Internal Free Heap 221700
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 65536
PlatformIO Built Environment: esp32doit-devkit-v1
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 80C40A24
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 267880, Internal Free Heap 236856
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      80C40A24
[NVS][Global][Boot Mode]:        3
[NVS][Global][STA SSID]:         Sauron2
[NVS][Global][STA Password]:     mypassword
[NVS][Global][AP SSID]:          Pedalino-80C40A24
[NVS][Global][AP Password]:      80C40A24
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    80C40A24
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Ladder]:           Ladder 1 Level 352
[NVS][Global][Ladder]:           Ladder 2 Level 533
[NVS][Global][Ladder]:           Ladder 3 Level 640
[NVS][Global][Ladder]:           Ladder 4 Level 723
[NVS][Global][Ladder]:           Ladder 5 Level 806
[NVS][Global][Ladder]:           Ladder 6 Level 908

Boot WIFI
USB MIDI started
DIN MIDI started
Connecting to
SSID        : Sauron2
Password    : mypassword
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
[... 129 lines cut ...]
Loading profile ...
Reading NVS Profile Adone
Bank  1
Pedal  1     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+   CONTROL_CHANGE       4   Channel  1   Pin D25
Pedal  2     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+   CONTROL_CHANGE       5   Channel  1   Pin D26
Pedal  3     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+      Channel  1   Pin D27
Pedal  4     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+   CONTROL_CHANGE      68   Channel  1   Pin D14
Pedal  5     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+   PROGRAM_CHANGE_DEC   0   Channel  1   Pin D12
Pedal  6     MIDI                MIDI         MOMENTARY1   PRESS_1_L     POLARITY+      Channel  1   Pin D13
Pedal  7     BANK+1    BANK-1    INFO                      PRESS_1_2_L   POLARITY+      Channel  0
MTC None

```

I think this is the problem

```
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
Update Start: firmware.bin
Update start fail: Partition Could Not be Found
Update fail: Partition Could Not be Found
SYSTEM_EVENT_STA_STOP
SYSTEM_EVENT_STA_STOP
ets Jun  8 2016 00:22:57
```

Seaway10, 2021-02-27:
I also try to update to HTTP interface but Update fail! and ESP restarted so what's the problem could be ?

alf45tar, 2021-02-27:
Comment line 151 in platformio.ini
Upload file system image
Upload firmware

Tomorrow I will fix the file

closed 2021-03-01 by alf45tar

teopost, 2021-03-01:
Tested. It works