alf45tar/PedalinoMini #192: Bank selection using switch

teopost, 2021-02-27:
Is there a way to assign a bank to a switch?

alf45tar, 2021-02-27:
Yes, there is.
Use Bank+ or Bank- and set Value = From Value = To Value = bank number.

teopost, 2021-02-27:
It would be very useful for me to be able to do this also in the configuration of the pedals. 

[image]

teopost, 2021-02-27:
Ignore my request.
I'm not sure it would be enough for me.
When I change bank I also have to change presets in HX Stomp and therefore I have to send a Program Change command.

teopost, 2021-02-27:
I would like to assign a bank to each switch.
Use case:

1. Long press to switch 2 to enable "Dani California" (Bank 2)
3. When Bank 2 is selected, I should send Program Change 3 to HX Stomp to select the right preset
4. The Bank 2 is selected, and now I can use all the switches of my pedalboard to send specifics midi signals for this preset.

Now, to obtain this, I have to configure the switch for preset selections, for each bank.
Maybe, It may be useful to have a special "Global" bank that works for all presets

alf45tar, 2021-02-27:
Global functions are defined into Pedals.
Using a dedicated bank is not a bad idea too.

mknerr, 2021-02-27:
Could you send the program change on press, and change the bank on release? Sorry, not able to test it at the moment.

alf45tar, 2021-03-01:
Global bank (also called Bank 0) has been added in the latest version. Actions in bank 0 will be triggered before the actions in the current bank.

Thanks again @teopost for the suggestion

closed 2021-03-01 by alf45tar