alf45tar/PedalinoMini #193: Long press and single press priority

teopost, 2021-02-28:
Button 4 in this image has 2 functions

[image]

1. LONG PRESS: TUNER
2. PRESS: Preset selection of my controlled device

When I do a "long press" the tuner is activated correctly,
but the press event is also triggered and the preset is changed

Is there a way to prevent the preset from being changed when I do a long press?

akosbeke, 2021-02-28:
HI @teopost 

Try changing the PRESS action to a CLICK action.

teopost, 2021-02-28:
The result is the same
When I activate the tuner, I don't want the preset to be changed

alf45tar, 2021-02-28:
Use CLICK for preset and LONG PRESS for tuner. It should work.

The event PRESS is triggered immediately on press and if the pedal is keep pressed for a long period also LONG PRESS is triggered.
CLICK is triggered when the pedal is pressed and released quickly before the LONG PRESS timeout.

teopost, 2021-02-28:
OK. I do some tests
Thanks for your patience!

teopost, 2021-03-01:
It works. Thank you

closed 2021-03-01 by teopost