alf45tar/PedalinoMini #194: Program Change+/- should start from last Program Change

teopost, 2021-03-01:
Yesterday I played with my band and my new Pedalinomini!
I still have to finish the configuration of Pedalinomini, but everything worked fine.
However, I have noticed a problem with this configuration:

* Button 4: "Program Change 1" - (Go to Preset 1)
* Button 5: "Program Change -"
* Button 6: "Program Change +"

During the performance I changed the preset with button 6 until reaching value 7

Then I pressed button 1 to go back to preset 1

When I later pressed button 6 I expected PC 2, instead Pedalinomini sent PC8

alf45tar, 2021-03-01:
Fixed in the latest version just released
Thanks for debugging

teopost, 2021-03-01:
Thanks @alf45tar 
Note: Using Pedalinomini with the band, I realized that the display is a bit small for me

alf45tar, 2021-03-01:
I agree display is a bit small but the project must remain affordable for everyone.

teopost, 2021-03-01:
Tested. It's works

closed 2021-03-01 by teopost