alf45tar/PedalinoMini #195: Schematic Pedalino mini

Seaway10, 2021-03-02:
Good time of a day ! I trying to use that schem https://github.com/alf45tar/PedalinoMini/blob/4c22f4098ad92a779bf36ae70132b56bc33ec30d/images/PedalinoMini_bb.png
there is a 9 buttons on electrical schem but in web pannel only 7 pedals which may to programming ( thats a strange). I have connected all wires like was shown . All 6 pedals seted up in  mode Momentory, Single press Actions, pedal 7 in default mode Momentory, single press Bank +, Double press Bank -. 
On the page Actions activated all 7 pedals and has applyed to each his value from 1 to 7 . When I click to  button 1 I see value 1 , click to button 2 is not responding , to button 3 I see value 1 but shoulb be 3,
to button 4 is not responding as well , button 5 value 2 but shoul be 5,
button 6 value 3 but shoul be 6 , so it's may be that schematic for pedalino is not correct ? 
And by the way in demo movie looks like other schematic when  that one  of  .png format.
Could you explane what is wrong ?

alf45tar, 2021-03-03:
M1, M1b and M1c are connected to Pedal 1 as Momentary 3.
M2 and M2b are connected to Pedal 2 as Momentary 2.
M3 to Pedal 3 as Momentary.
M4 to Pedal 4 as Momentary.
M5 to Pedal 5 as Momentary.
M6 to Pedal 6 as Momentary.

For Pedal 1 configure actions on Button 1 , 2 and 3.
For Pedal 2 configure actions on Button 1 and 2.
For Pedal 3, 4, 5 and 6 actions on Button 1.

Seaway10, 2021-03-03:
> M1, M1b and M1c are connected to Pedal 1 as Momentary 3.
> M2 and M2b are connected to Pedal 2 as Momentary 2.
> M3 to Pedal 3 as Momentary.
> M4 to Pedal 4 as Momentary.
> M5 to Pedal 5 as Momentary.
> M6 to [...]

That's Great  thank you alot ,everything are works so what about Pedal 7 ?

alf45tar, 2021-03-03:
Pedal greater than 6 are on-board pedals.
Pedal 7 is the BOOT button on ESP32 board.

closed 2021-04-01 by alf45tar