alf45tar/PedalinoMini #197: Hide Bank 0  from Bank+/- navigation

teopost, 2021-03-06:
From Bank 1 I decrease the Bank position (Bank- function) and Pedalinomini go to Bank 0
Bank 0 is a special bank that contains global actions and I think should be skipper from Bank navigation.

I also verified that:

* From Bank 0 Bank- go to Bank 20
* From Bank 20 Bank+ remain in Bank 20

alf45tar, 2021-03-06:
Set From Value and To Value to limit the range of Bank+ and Bank-

teopost, 2021-03-06:
Good advice. Thank you!.
Yes. I think (It might be useful to add the last commit number in the PedalinoMini information)
The last change you made (Global bank) was great.

alf45tar, 2021-03-06:
Thanks to you for the idea.

alf45tar, 2021-03-06:
Firmware version is on the homepage of the WebUI. Currently is 2.1.0.

teopost, 2021-03-06:
I still have a few issues I'm investigating!
I want to configure a button that always activates bank 1.

teopost, 2021-03-06:
We are almost there

https://youtu.be/TFaCFuChZlY

teopost, 2021-03-06:
there is a bug that I cannot systematically replicate
pedal 4 is configured like this 

[image]

When I click it sends program change 1 and it has to go to bank 1 also
I have to do this because sometimes the sync between the bank and the preset is lost (I don't know the reason)

well it doesn't work.

watch the video 

https://www.youtube.com/watch?v=wuhmTJ8Dwu4

teopost, 2021-03-06:
Sorry. The problem is another.
The Bank name  (DANI) is changed!
I should investigate. Please ignore

alf45tar, 2021-03-06:
You are pushing Pedalino over its possibilities. I appreciate your hard work.

closed 2021-03-14 by teopost