alf45tar/PedalinoMini #199: By pressing the buttons nothing happens

davidin73, 2021-03-13:
Hi!
My problem is that if i push some already programmed buttons, on the screen nothing happens, and in the terminal is seen only one of the two buttons. This is for all the buttons

alf45tar, 2021-03-13:
Hi @davidin73 
I have not enough details to troubleshooting your issue. Post the log and the schematics of your pedal connections.

davidin73, 2021-03-13:
Hi @alf45tar i have attached a video in my comment....
The pedalboard is based on the @marosell project, the buttons are set as actually 3 momentary2 buttons

alf45tar, 2021-03-13:
Your buttons are latches not momentary buttons.
Configure the pedals as Latch 2 and the events to use are Press for actions 1 and Release for action 2.

davidin73, 2021-03-13:
Thank you very much.
When I come at home i do a test and the result.

davidin73, 2021-03-13:
Hi, I'm here...
Nothing to do , i haven't solved.
I have connected the buttons as in the boss fs-6 schematic type,if this can help....

davidin73, 2021-03-20:
Ok, finally the board is broken....
I return on your standard esp32 board.
Thanks for your help

closed 2021-03-20 by davidin73