alf45tar/PedalinoMini #203: config files structure

rigr, 2021-03-23:
Thanks again alf45tar for your great project. I just installed the latest version and all looks good after some minor tweaks (the options-page was not displayed, it took several reboots).
While setting up the pedalino I thought it would be cool if I could use an external (java) program to write a config file and then upload, because clicking through the interface is very time consuming and would be a lot easier to be done on my laptop. Such a software might even be enhanced with graphical environment to show colors of LEDs etc.
Would you mind explaining the config files?
Thank you and best regards
Richard

alf45tar, 2021-03-23:
Before any update save configuration, reset to factory default and reload configuration. EEPROM can change without any further advice.

There is nothing fancy in the configuration file. It is just a JSON file.

{
  "Pedals": [
    {
      "Pedal": 1,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 2,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 0,
      "Max": 127
    },
    {
      "Pedal": 2,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 2,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 0,
      "Max": 127
    },
    {
      "Pedal": 3,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 2,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 0,
      "Max": 127
    },
    {
      "Pedal": 4,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 2,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 0,
      "Max": 127
    },
    {
      "Pedal": 5,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 4,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 102,
      "Max": 921
    },
    {
      "Pedal": 6,
      "Function": 1,
      "AutoSensing": 0,
      "Mode": 4,
      "PressMode": 1,
      "InvertPolarity": 0,
      "MapFunction": 0,
      "Min": 102,
[... 303 lines cut ...]
      "Bank": 2,
      "Pedal": 5,
      "Button": 1,
      "Name": "[ E ]",
      "Event": 6,
      "Message": 176,
      "Channel": 1,
      "Code": 12,
      "Value1": 0,
      "Value2": 127
    },
    {
      "Bank": 2,
      "Pedal": 6,
      "Button": 1,
      "Name": "[ F ]",
      "Event": 6,
      "Message": 176,
      "Channel": 1,
      "Code": 13,
      "Value1": 0,
      "Value2": 127
    }
  ]
}

rigr, 2021-03-24:
Thank you, so you are not parsing it for line feeds and spaces.. I'll hope to be able to play around with uploading files to my pedalino after editing them with notepad++

Thank you an best regards
Richard

alf45tar, 2021-03-27:
The Configuration Editor has been added to Configuration page. Give a try.

rigr, 2021-03-27:
wow, this is great! will play around asap.
Thank you a lot!

rigr, 2021-03-27:
It works great, thank you!
Please allow me to suggest you set a link to
https://github.com/alf45tar/PedalinoMini in the headerline of the
pedalino-webinterface.

Best wishes
Richard

closed 2021-03-30 by alf45tar