alf45tar/PedalinoMini #204: Optical type expression pedals in line 6 fbv boards

jimhiggs, 2021-03-24:
I recently picked up a Line 6 fbv longboard and a shortboard mk11 for not a lot of money on kijiji here in Canada. I am contemplating the options open to me for converting one of these to pedalinomini to control my Helix Stomp.
I currently have a little black pedalinomini box with sockets for external pedals and it all works great, thinking it would be nice to have it all in one pedal box.  I thought the shortboard mk11 would be able to drive my Stomp over USB midi but have discovered it needs a host in between. Also thinking the original shortboard may be a better candidate for conversion as it has better switches. 

My issue, all the fbv pedalboards use Optical (shark fin) style expression pedals.
Is there a way these could be used into one of the pedalinomini inputs?

alf45tar, 2021-03-24:
Optical encoder should be configured as jog wheel.

jimhiggs, 2021-03-24:
But does the Jog wheel input just detect the movement and direction of the jog wheel? For an expression pedal it would need to detect the current position. I think they work by reducing/increasing the light falling on the optical sensor but you probably already know that.

renne, 2021-12-18:
@jimhiggs Most optical encoders use a gearbox and a slot in a disc creates impulses. They can have some kind of mechanism which creates a reset signal when the min or max position is reached.

The first question would be if your slider uses a real optical distance sensor (quite rare) or just a slot creating impulses.

jimhiggs, 2021-12-18:
@renne Both the ones I have taken apart have a strip of plastic attached to the foot pedal that is shaded by the look of it, gradually darkening along its length and that is passed between what I assumed was an optical sender and receiver.

renne, 2021-12-19:
@jimhiggs 
Are there any part numbers on the optical sender/receiver?
Is there a additional circuit?

jimhiggs, 2021-12-20:
@renne I had given up on it put it back together. Will have to dig it out and take a look.