alf45tar/PedalinoMini #206: Buttons stopped working

johnpaularthur, 2021-03-25:
This is an odd issue.   My project was working a few days ago.  Today I added a new encoder/switch.   I could have burnt something out unwillingly.   But I don't think that is the issue.   I tried a system reset to no avail.  However, I can still use the ladder config on boot and have the buttons respond during that setup.  But after boot, nothing happens.   Even the "Button 7" on the board doesn't work as a momentary button.  To be fair, I've never tried the on board "EN" button before so I can't say if it worked previously.

Anyways, I can reflash the board, but I thought this was an interesting issue and would like to know if there is a cause.

johnpaularthur, 2021-03-25:
If it matters, I just updated to 2.1.3 with the same issue (was 2.1.2).  I think now it might have been related to 2.1.2/3.  I can not configure the switches.  They start with the inverted option enabled and no matter what I do (enable single press, etc), each switch reverts to inverted and no other options enabled.  All i can do is enable or disable inverted.

johnpaularthur, 2021-03-25:
Ok, so I tried flashing backwards to 2.1.1.   Fresh setup.  Before setting up ladder config, I could change the switch options like normal.  But once I went thru the ladder config, the pedal config options (single, double , long, etc) stopped working and won't take a change.   
So I erased the memory (esptool) and started with a fresh build of the most recent 2.1.3.  Still had the same issues.   So I flashed 2.1.0 and everything works now.   Something seems to be off about the new pedal UI in version > 2.1.0.

alf45tar, 2021-03-26:
>Before any update save configuration, reset to factory default and reload configuration. EEPROM can change without any further advice.

Reset to factory default because EEPROM is changed.

johnpaularthur, 2021-03-26:
Yes, I reset to factory default each time.   Now that I'm back on 2.1.0 and running successfully for the last day.  I'll try the updates again and see if I face the same issue.

alf45tar, 2021-03-26:
Fix will be released soon

alf45tar, 2021-03-26:
Version 2.1.4 is out!

johnpaularthur, 2021-03-27:
Perfect.  Things are working well now.

closed 2021-03-29 by alf45tar