alf45tar/PedalinoMini #208: Enhancement request:  OLED display per switch (example project reference)

johnpaularthur, 2021-03-29:
I'd love to see each switch control it's own OLED screen like say the HELIX.  Mainly do to the limited screen size we have now.   It looks to be possible with the ESP32 using  SSD1306/SSH1106 I2C.   Here is someone who has worked on an example project: https://bitbanksoftware.blogspot.com/2019/01/controlling-lots-of-oled-displays-with.html

DWSQUIRES, 2021-06-22:
Just adding to your request. More or larger screens would definitely be a huge benefit. 
Maybe an idea here? very low cost 8 way I2C expansion
https://www.youtube.com/watch?v=vV42fCpmCFg&ab_channel=DroneBotWorkshop
https://www.aliexpress.com/item/4000067621113.html?spm=a2g0o.store_pc_allProduct.8148356.2.545526f1iEurIi

A7F, 2021-08-13:
this one looks similar to #230 
maybe that multiplexer is a good choice specially considering the case where someone wants to make a pcb...

DWSQUIRES, 2021-08-13:
I think it would be a great addition, the parts are so cheap that even 1 OLED per switch doesn't add much to the total project