alf45tar/PedalinoMini #209: Problem in configuration backup

teopost, 2021-04-05:
I still haven't understand out why but all bank 0 configurations have been lost!

The problem is that the copy I had saved does not have the Actions of bank 0

[image]

The first bank saved is the bank 1. 

[image]

The last is the bank 14

I'm very sad tonight! I lost all my work!

alf45tar, 2021-04-06:
Let see if we can recover your configuration file.
First of all which version are you using? Configuration file is changed from 2.1.6 in order to be more user friendly when edited in the configuration editor.
Can you attach your configuration file? I can have a look and fix it for you.

Cheers
alf45tar

teopost, 2021-04-06:
I have an update.
The settings have not been lost, I found them all in bank 1
They have moved from bank 0 to bank 1
I think there is a problem in the interface but I still haven't been able to replicate it

teopost, 2021-04-06:
The version I use is 2.1.5
The last Cloud version available is the 2.1.4

[image]

teopost, 2021-04-06:
I did a git pull and loaded the latest firmware version
Now pedalinomini reboots continuously

This is the log

````
Boot WIFI
USB MIDI started
DIN MIDI started
Connecting to
SSID        : Sauron2
Password    : *******
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
BSSID       : 5C:49:79:47:93:75
RSSI        : -58 dBm
Channel     : 6
STA         : 24:0A:C4:80:4E:90
SYSTEM_EVENT_STA_GOT_IP
Hostname    : 80C40A24
IP address  : 192.168.178.47
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.178.1
DNS 1       : 192.168.178.1
DNS 2       : 0.0.0.0
mDNS responder started
OTA update started
HTTP server started
Connect to http://80C40A24.local/update for firmware update
Connect to http://80C40A24.local for configuration
ipMIDI started
RTP-MIDI started
OSC server started
Internal Total Heap 270132, Internal Free Heap 143592
Loading profile ...
Reading NVS Profile Adone
Bank  2
Pedal  1     1                PRESS_1       POLARITY+      Channel  0
Pedal  2     1                PRESS_1       POLARITY+      Channel  0
Pedal  3     1                PRESS_1       POLARITY+      Channel  1
Pedal  4     1                PRESS_1       POLARITY+      Channel  0
Pedal  5     1                PRESS_1       POLARITY+      Channel  0
Pedal  6     1                PRESS_1       POLARITY+      Channel  0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x801380d4  A1      : 0x3ffcfde0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x3ffd0213  A11     : 0x3ffcffe3  A12     : 0x3ffd0214  A13     : 0x00000001  
A14     : 0x00060723  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffcfde0 0x401380d1:0x3ffcfdf0 0x4013d75a:0x3ffd0100 0x4013d796:0x3ffd0190 0x40126776:0x3ffd01d0 0x400fc2ae:0x3ffd0270 0x400fc1bd:0x3ffd02d0 0x400fe055:0x3ffd0330 0x40129dc1:0x3ffd0380 0x40091a0a:0x3ffd03a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
```

teopost, 2021-04-06:
Hi,
  this is my configuration file

[HX Stomp conf.cfg.zip](https://github.com/alf45tar/PedalinoMini/files/6266573/HX.Stomp.conf.cfg.zip)

teopost, 2021-04-06:
No problem!
I can recreate the configurations
Thanks for the support

closed 2021-04-07 by teopost