alf45tar/PedalinoMini #210: Bank 1 is selected when changing to profile A

teopost, 2021-04-05:
I have configured a button to switch profiles

[image]

When I switch to the profile B and C, bank 0 is selected.
When I switch to the profile A, bank 1 is selected
This is the sequence
A01 -> B00 -> C00 --> A01 ecc..

alf45tar, 2021-04-06:
The logic is to have a default bank by profile and not to store the last bank used in any profile.
Current bank is saved within the profile when user saves the profile via WebUI not when bank change by pressing pedals.
Does it makes sense?

teopost, 2021-04-07:
I think that bank 0 is a special kind of bank.
I believe bank 1 should be always the default bank for every profiles

alf45tar, 2021-04-08:
Yes, the default bank is 1 if not changed by user

https://github.com/alf45tar/PedalinoMini/blob/fdd0220ae94ff4595863ab5783f903a7f39b2d16/src/Config.h#L569

closed 2021-05-01 by alf45tar