alf45tar/PedalinoMini #212: Web interface display problems.

SD456, 2021-04-17:
Hello everyone!

I have successfully uploaded PedalinoMini to my TTGO T-Display, but the Web interface is not displayed correctly. I have tried in Chrome, Firefox, and Microsoft Edge. I have disabled the browsers extensions too, but it did not changed anything.

Thanks for the help in advance!

Images:

[image]
[image]
[image]

rigr, 2021-04-17:
I had the same "problem" and solved it in "Options" changing the
web-UI-theme.

alf45tar, 2021-04-17:
The solution is in the session 'How to build your Pedalino' and reported here for your earlier convenience.

Click “Upload File System Image” under Platform. Do not skip this step otherwise the WebUI will not works properly.

rigr, 2021-04-17:
It once also showed because I did not reset to defaults before starting an
upload.

SD456, 2021-04-17:
> The solution is in the session 'How to build and upload' and reported here for your earlier convenience.
> 
> Click “Upload File System Image” under Platform. Do not skip this step otherwise the WebUI will not works properly.

My bad, I was not paying attention, thanks for the info!

closed 2021-04-17 by SD456