alf45tar/PedalinoMini #214: Green box built!

gallochri, 2021-04-27:
Hi, my first pedalino.

[image]

rigr, 2021-04-27:
wow, looks great! What box did you use? And maybe install some tempered glass over the TTGO.
Regards
Richard

alf45tar, 2021-04-27:
Great job @gallochri

alf45tar, 2021-04-27:
To simplify the enclosure build process I am working on

https://github.com/alf45tar/PedalinoMini-Case-1

Dante777, 2021-04-27:
Hi, great box, @gallochri! Are these LEDs connected with button state? If so, could you, please, share the connection scheme?

gallochri, 2021-04-27:
> wow, looks great! What box did you use? And maybe install some tempered glass over the TTGO.
> Regards
> Richard

https://catalog.weidmueller.com/catalog/Start.do?ObjectID=group23308260189911
I am an electrician  ;-)

gallochri, 2021-04-27:
> Hi, great box, @gallochri! Are these LEDs connected with button state? If so, could you, please, share the connection scheme?

I used the schema in the readme file.
https://github.com/alf45tar/PedalinoMini/blob/master/images/Schematic_PedalinoMini_Sheet-2.svg

gallochri, 2021-04-27:
Other photos...
[image]
[image]
[image]

teopost, 2021-04-28:
Wow!  another picture for the github home page!

davidin73, 2021-04-28:
> To simplify the enclosure build process I am working on
> 
> https://github.com/alf45tar/PedalinoMini-Case-1

Seems nice!!!!!

teopost, 2021-07-27:
@gallochri, what happens if a glass of beer falls on the pedaboardl?  :-)

gallochri, 2021-07-27:
Never drink & play ;)

closed 2021-07-29 by alf45tar