alf45tar/PedalinoMini #215: How to wire up six 3PDT switches to the T-Display board?

SD456, 2021-05-05:
Hello,

I would like to ask a basic question, how should I wire six 3PDT switches to be able to use as midi controller? I have a TTGO T-Display board.

Thanks for the help!

alf45tar, 2021-05-07:
Connect one pole (the external row of pins) to a digital pin (D1, D2,...) and the central pin to GND.

SD456, 2021-05-07:
Thanks for the reply. Should I use ladder configuration or just connect them to the board pins?

Also, I would like to use an external expression pedal too.

alf45tar, 2021-05-07:
Connect the first to D1, the second to A1, the third to D2, the forth to A2 and so on.
Configure the first 3 pedals as Momentary 2 and you still have 3 ports available.

SD456, 2021-05-07:
> Connect the first to D1, the second to A1, the third to D2, the forth to A2 and so on.
> Configure the first 3 pedals as Momentary 2 and you still have 3 ports available.

Okay, thanks for the help, I will try this!

closed 2021-06-07 by alf45tar