alf45tar/PedalinoMini #219: I am using B10k pot and I see a lot of jumping around in the output

peeyush-shippo, 2021-05-14:
Hi I am using a B10K pot hooked upto A2,D2. Its working except I don't see full range and output is not stable I see a lot of jumping around. I am using auto calibration. The schematic I saw here was using Yamaha which has 50K pot. would you recommend trying a 50k pot?

alf45tar, 2021-05-14:
DId you connected the GND? Did you solder the pot or used a breadboard? Breadboard loose connection can be the reason.
10k or 50k are ok both.

akosbeke, 2021-05-14:
Are you using power from your computer or a dedicated power supply? Jumping like this can happen due to the lack of stable power.

peeyush-shippo, 2021-05-14:
I am using power from my computer. let me try using dedicated power supply.

closed 2021-07-27 by alf45tar