alf45tar/PedalinoMini #220: Boot Bouton and GPIO0 on 38pins board not working to select profiles

juani13973, 2021-05-14:
Hi alf45tar,
I am almost there....
I have a ESP32 DOIT with 38 Pins and accessories to the GPIO0 to externalize the Boot Bouton.
Now, the problem I have after uploading to the sponsor version is the following :

The boot bouton is not working after booting. When I boot I can select one of the 7 modes, but after board is up and running, the ESP32 board Boot Bouton is not working to select these options:

[image]

Finally when I had the old software, I could pull down the GPIO0 and externalize the Boot function. Not it doesn't work either....

Any reasons for this ?

The rest is working super ! - thanks for your great contribution to the world of music. A quick pic of how my prototype is looking... (I need to build the box to plug the front panel in 3D and I am done... by the way I have 3 front panels, happy to share if someone can print me the base.... )

[image]

Thanks a lot for your support and the answer on the boot issue !!!

alf45tar, 2021-05-14:
Awesome. It is the world most huge PedalinoMini ever built.
Look forward to share details and photos.

alf45tar, 2021-05-14:
You are right the BOOT button is not working as reported in the documentation.
There is a workaround: create manually the actions for pedal 7.
Il will fix soon but it will not work for you because you are using a custon configuration.

juani13973, 2021-05-15:
Thx alt45tar, what do you mean I am using a custom config ?
Regarding the workaround, I thought that the pedal 7 was actually the boot bouton with the single, double and long press options.

alf45tar, 2021-05-15:
Do you know the new global bank 0? Every actions defined in the bank 0 are always active.
Pedal 7 is the BOOT button and there are no any hidden actions. If you want to use for profile+/profile- define actions in bank 0 in avery profile. These actions are usually added in the factory default configuration.

closed 2021-07-27 by alf45tar