alf45tar/PedalinoMini #221: Battery Status in "esp32dev"

TarFilarek, 2021-05-15:
Hello!

Do You have any plans to implement the "Battery Status" option in ordinary ESP32 without a battery connector?

In the older releases, it was implemented with the usage of the voltage divider and one GPIO (32 to 39).

I've tried to add it manually.

```
build_flags_esp32 =
-D PLATFORMIO_ENV=$PIOENV
-D PEDALINO_MINI
-D DEBUG_ESP_PORT=Serial
-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
; -D SSD1306WIRE
-D SSH1106WIRE
-D WEBSOCKET
-D DIAGNOSTIC
; -D NOSMARTCONFIG
; -D NOWPS
; -D NOWEBCONFIG
**-D BATTERY**
```

After uploading, it shows an error that "BATTERY_ADC_EN" is not defined.

```
#ifdef BATTERY
pinMode(BATTERY_ADC_EN, OUTPUT);
digitalWrite(BATTERY_ADC_EN, HIGH);
delay(5);
uint16_t v = analogRead(BATTERY_PIN);
digitalWrite(BATTERY_ADC_EN, LOW);
```

"BATTERY_ADC_EN" is defined only in **HELTEC_WIFI_KIT_32** and **TTGO_T_DISPLAY**. In other cases, only the "BATTERY_PIN" is defined.

I've tried to change "BATTERY_ADC_EN" into "BATTERY_PIN" in the above code, and the upload was successful, but the battery indicator didn't show up.

Kind Regards

alf45tar, 2021-05-16:
There is no need to enable battery reading on a generic esp32 because the circuit is not present. I suggest to delete the lines 

>pinMode(BATTERY_ADC_EN, OUTPUT);
>digitalWrite(BATTERY_ADC_EN, HIGH);
>delay(5);

Battery status could be implemented/fixed also for a generic esp32 board but some components must be external.

alf45tar, 2021-05-16:
Fixed in sponsor version

closed 2021-05-17 by TarFilarek

TarFilarek, 2021-05-27:
> some components must be external

Previously (version 2.0), I've used a voltage divider as described, shown and implemented below...

> As an example, for a single cell Li-Ion battery (4.2V - 3.7V) powering a 3.3V MCU, you'll need to use a voltage divider with a ratio no less than 1.3. Considering only E6 resistors, you can use a 4.7kΩ (R1) and a 10kΩ (R2) to set a ratio of 1.47: this allows to measure batteries with a maximum voltage of 4.85V, well within the swing of a Li-Ion. 

Schematic

[image]

Code from the old version

[image]

... and it worked quite well.

Should I add something else? Any other external component or additional code line?

I've checked the sponsors-only version of the code and it solved my previous problem, but the battery indicator did not show up. I've noticed, that almost all code lines connected to the battery status indicator are greyed out and labeled as undefined "BATTERY2". 

[image]

[image]

Do You have any idea how to solve this problem and unlock the Battery Status Indicator for "esp32dev"?
I will totally understand if You don't have time to deal with this feature right now. 

Kind Regards

alf45tar, 2021-05-27:
Battery status has been abandoned on esp32dev because the generic board doesn't have enough hardware.
Adding a couple of external resistors is not usually enough because they will continuously discharge the battery.
Board like TTGO T-Display implement the voltage divider and a "switch" to connect battery to voltage divider just for the time of voltage reading.
The best I can do for you is to fix the software but I cannot solve the battery discharge if you do not implement the "switch". My suggestion is to move to TTGO T-Display.

TarFilarek, 2021-05-27:
I understand. I would really like to switch to TTGO, but it would be extremely difficult to modify my current setup. I've put quite a lot of work into that, and it would be pretty painful to abandon it without even finishing it.

I'm using four 18650 3.7V cells, so it lasts pretty long. Adding a switch wouldn't be a problem either. I already did that for the MIDI DIN Diodes circuit.

I don't know if this fix is time-consuming, but it would mean the world to me. If it is a problem, then don't worry - I totally understand and I don't want to be a problem for You.

[image]

alf45tar, 2021-05-27:
Really nice and clean job.
No problem I will fix the software.

alf45tar, 2021-05-27:
Posted the new code. Remember to define BATTERY in platformio.ini because is disabled by default.

TarFilarek, 2021-05-28:
It works perfectly!
You have no idea how thankful I am right now. You saved me from a lot of frustration. Thank You!

https://user-images.githubusercontent.com/64553282/120032752-f0f13180-bffa-11eb-8eda-a86ce36d7a51.mp4

> Connect the BATTERY_PIN not to battery output but to VIN (the charging circuit output).

I'm using a battery shield with a constant output voltage of 5.12 V, so the battery indicator based on values from that output would constantly show "full battery" / "charging" till the very end, then the battery would simply die out. Because of that, I had to connect to the plus side of the first cell in the circuit, to read the real cell voltage. The value shown in Pedalino Info is almost identical to the multimeter reading of this cell (± 0,03V). All four cells are discharged almost simultaneously (the biggest difference at about ± 0,15V), so this solution works quite fine. 

Thank You very much!

wybbbb, 2021-09-06:
> 我明白。我真的很想切换到 TTGO，但是修改我当前的设置非常困难。我为此付出了很多努力，甚至没有完成就放弃它会非常痛苦。
> 
> 我使用了四节 18650 3.7V 电池，所以它可以持续很长时间。添加开关也不是问题。我已经为 MIDI DIN 二极管电路做到了这一点。
> 
> 我不知道这个修复是否耗时，但这对我来说意味着整个世界。如果这是一个问题，那么别担心——我完全理解，我不想成为你的问题。
> 
> [image]

Hello, can you share the production file of the shell?