alf45tar/PedalinoMini #222: OTA Update for SSH1106WIRE display

teopost, 2021-05-16:
After the OTA update my display is not working properly.
The display of my Pedalinomini is SSH1106WIRE.
The version I am using was previously compiled by me modified the file platformio.ini

teopost, 2021-05-16:
ADDENDUM:
II reloaded the previous firmware 2.2.1, but now the WEB UI is like this

[image]

alf45tar, 2021-05-16:
Version 2.2.2 include a new Bootstrap version. The corresponding System Image must be uploaded.

alf45tar, 2021-05-16:
I suggest to apply the same changes you did to platform.ini to the new version 2.2.2.

teopost, 2021-05-16:
can I do it via OTA?

alf45tar, 2021-05-16:
Yes, you can. Compile using platformio ide, goto .pio/building looking for firmare.bin and spifferi.bin file, upload them connecting to http://<pedalino ip or name>.local/update

teopost, 2021-05-16:
Thanks you. It's works

closed 2021-05-16 by teopost