alf45tar/PedalinoMini #224: Defective esp32 board?

davidin73, 2021-05-18:
Hello
After various attempts to send midi commands to my Helix', and various controls, i think my esp32 board is defective in the the midi out section.
How can I verify if is it?
Thanks in advice

alf45tar, 2021-05-18:
There is no midi out session in the esp32. It is just a serial interface. 
Which midi out schematic did you used? Did you add the led? Is it blinking?

davidin73, 2021-05-19:
Hello.
No ,without led....i can modify it this morning to add the led.
I have used the standard schematic ....

davidin73, 2021-05-19:
Done the LED's add....
Is flashing on pushbutton

alf45tar, 2021-05-19:
It is a good news. Are you sure of connection? Try to invert pin 4 and 5 on MIDI connector. It is not dangerous because there is a diode in every MIDI IN port to protect against pin inversion.

davidin73, 2021-05-19:
Hm....i think I have failed the resistences values on the midi port and the +5v midi connector...🤦🏻‍♂️
I have re measured, and aren't 220 ohms,but around 190 ohms....
the sight begins to play tricks on me ... 😞i try to replace them, and later i post you the result...
Thanks for your patience

davidin73, 2021-05-19:
Solved
Effectively was the two cables inverted in the midi connector
Thanks for your patience!!!!

closed 2021-05-19 by davidin73