alf45tar/PedalinoMini #228: Any plans for supporting SPI connected displays?

MiqViq, 2021-05-29:
I recently purchased a bunch of different electronics parts for my upcoming projects.
For testing purposes I got also a few 0.96" white OLED displays that are using SPI.
I guess that code does not support SPI displays (yet), are you planning to include SPI support?

alf45tar, 2021-05-29:
Can you share a link of the OLED?
Every display supported by https://github.com/ThingPulse/esp8266-oled-ssd1306 or by https://github.com/Bodmer/TFT_eSPI could be used.

MiqViq, 2021-05-29:
The white 0.96" with SSD1306 7-pin spi here:
https://www.aliexpress.com/item/32638662748.html

Is there any need to modify the source code and create a new custom firmware for such SPI display support?

I have also this with SH1106 (seems to be 132x64) SPI but I have not found any library that supports it:
https://www.aliexpress.com/item/32683094040.html
That 132x64 is not so common format so it is probably not the best display to use...

alf45tar, 2021-05-30:
Both 4 pins version are totally supported.
Why using 7 pins (SPI) when 4 pins (I2C) is enough? We don't have too much spare pins on ESP32.

closed 2021-07-27 by alf45tar