alf45tar/PedalinoMini #229: My final prototype juani13973 -> Thanks alf45tar

juani13973, 2021-05-29:
Dear alf45tar,
Here below the pictures of my prototype. I really enjoy working on your project. I am an electronic engineer that hasn't touched a resistor in 25 years. I was looking for a footswtich and google brought me to your open source project. And I said... why not ?
I think I spent the same amount I would have spent if I bought it in a store but !! I LEARNT A LOT :
1. I discovered and learnt to use GitHub
2. I discovered what happened in the last 25 years with microcontrollers. I used to program in assembler.. (incredible)
3. I learnt somehow to use arduino and visual studio
4. I learnt to design in FUSION 360 !! that software is free !! - PCBs and the entire project.
5. I did PCBs again and instead of DIY at home, I learnt how to buy them in China. What a change !
6. I did my first 3D print ever. I was wondering what 3D printing is really for. I discovered a new world.
7. I discovered how cheap things are in AliExpress :) - no more amazon 

Now I am willing to continue exploring coding in my free time !!
I can share my Fusion 360 project and PCBs if someone is interested. I do not recommend to print the box in 3D. It is very expensive. I should have done a metal sheet metal folding. (I might do that later). The rest was cheap and the fun was great !!

Alf45tar, thanks again for bringing me back to UNI and motivate me on this. Amazing journey !! I think that I discovered some bugs or I do not know how to use some features but I will comment on other threads. I already used several days the footswitch and it works GREAT with BIASFX2, Anytune, Qunatiloop all running in my iPad.

Thanks !
juani13973

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

alf45tar, 2021-05-29:
You are a pro. Guinnes World Record as biggest PedalinoMino ever build.
Will to learn and passion guide us.

alf45tar, 2021-05-29:
@juani13973 if you want to share your files I suggest to create a github repository

MiqViq, 2021-05-29:
@juani13973 , seems like you have spent some time for designing the internals there...good work and thanks for sharing photos, very informational for others!
Can you share some links that were helpful for you when learning designer apps, like Fusion360 and such?
 
Shops in Aliexpress have cheap prizes but there are sometimes issues with deliveries, at least if you select the cheapest way of shipment. I am still waiting for a few packets that were ordered around 2020-11. Seems that it is best to use shipping types that have tracking possibility, like Aliexpress standard shipping. And the delivery time is only about 2-3 weeks, at least from China to Northern Europe.

MiqViq, 2021-05-29:
@juani13973 I see that your box is using Dupont connectors for cabling, as "Arduino, RasPi, Whatever" prototype builds usually have nowadays.

After I have verified that my parts and connections are working properly I am planning to use something like this, for more robust and "professional" connectors:
https://www.aliexpress.com/item/32954418743.html

My proto build may take some weeks to finish as I have other concerns right now...

juani13973, 2021-05-30:
Thx alf45tar for your comments.
MigViq, give me some time and I will create a repository with all docs and files I created. (Need to take care of the kids :) )

I agree on connectors. In my learning journey I found yours after I had already bought some staff and I didn't want buy again. Your recommendation is what needs to be used and they fit in my PCB as well.

Regarding Aliexpress, I never had an issue. The tracking system works perfect. I live in Paris, France. Some things took 3 to 4 weeks (the footswitch for example) but others were arriving almost the week after (insaine....). Everything I bought has arrived. I had 1 issue with banggood.com where I also bought components.

I will share later during the week.

juani13973, 2021-06-01:
Hi Miqviq
I created this repository with all the info I got about my prototype. Again thanks alf45tar !
https://github.com/juani13973/pedalinio-18-footswitches-prototype

MiqViq, 2021-06-01:
Good job @juani13973 .
I will check if I can import some of the files into Fusion360 later next week, maybe.
It will be a learning lesson for me too as I have not yet designed anything with Fusion360...

juani13973, 2021-06-01:
Going over the free training of Fusion 360 in the Autodesk site is enough to learn. Autodesk give you a hobby license for free that allow you to open up to 10 projects at the same time and designs PCBs of 80cm2 max. (That is why I had to do 3 footswitches PCBs).

Now to learn how to use a CAD, I would recommend if you have an iPad and the pencil to de the tutorials of [Sharp3D](https://www.shapr3d.com) to understand the concept of designing bodies in 3D. It is super intuitive.

But the app is not free and limited in functionality. Fusion 360 is 10 times more powerful but you need to use it in your laptop.
When you go to the training of Fusion, if you have the background of Sharp3D it will be easier to digest.

closed 2021-07-29 by alf45tar