alf45tar/PedalinoMini #230: Using an additional LCD/LED display for selectable info?

MiqViq, 2021-05-29:
As current OLED displays are quite small for distance viewing it would useful to have a LCD/LED display module that could show some useful info, like patch name.
Any plans to add support for that?
I guess that ESP32 could control at least 2 separate displays, especially if they are using a different type of connection?

alf45tar, 2021-05-30:
I do not have plan to support additional display because I want to keep it simple and there is no more pins to use.
If someone have a solution let me know.

MiqViq, 2021-06-01:
I2C bus can handle multiple displays by default.
There are numerous examples with code bits on the internet...here is one:
https://electronics-project-hub.com/how-to-connect-multiple-lcd-to-arduino/
What we need to do is to define a different address for the second display. This must be done by modifying jumper pins on the display's I2C module/board, I guess...