alf45tar/PedalinoMini #232: Page turner - sending keystrokes or converting midi messages into keystrokes

juani13973, 2021-06-05:
Hi to all,
I would like to use a PDF reader to turn pages with pedalino mini. Is there a way to send keystrokes with pedalino mini ?
if not, would you know of an IPAD app that converts midi messages to keystrokes ? (there is one for Mac [Bome](https://www.bome.com/products/miditranslator), but I use my iPad)
Or would you know a pdf reader that support midi commands ?
Thanks again !

alf45tar, 2021-06-06:
Did you have a look here https://forscore.co?

alf45tar, 2021-06-07:
I did some research and testing.
The good news is that it could be implemented quite strait forward using the following library https://github.com/T-vK/ESP32-BLE-Keyboard.
The bad news is we don't  have enough memory. To add this feature we should to remove OTA update.
Does it makes sense?

juani13973, 2021-06-07:
Thx alf45tar for keep looking. It would be interesting for the community. I think I will go with the **forscore app.**

I do not use the bluetooth connection. When I found the Pedalino mini project by chance, I was looking for a footswtich that had a wired interphase :) so I use the usbmidi connection. (I have some bluetooth foot switches but for x reason the disconnect randomly and I hate it when using the looper). And on top with usbmidi I leave everything connected all the time. I turn it on and it just works.

I never tried OTA but next time I will have to upgrade the firmware I will do, if not I need to open the footswitch so I prefer to keep that function.

Said so the keyboard function is very useful to control other apps such as YouTube, [Playsections](http://playsections.github.io/?setLng=en) or a music app etc. Most of the apps responds to keyboards ...

the ideal would be to have an app in IOS that translate CC midi commands into keystrokes ones.

thx again,

juani13973, 2021-06-07:
Another quick one ? Is there any reason why the usbmidi is not implemented in the esp32 directly ? Why would we need the promicro ?

alf45tar, 2021-06-08:
ESP32 microcontrollers don't have native USB support and development boards use a single-purpose USB-to-TTL chip,
Arduino Pro Micro is a cheap and compact microcontroller with native USB capabilities (atmega32u4 based boards) to appear as a MIDI peripheral over USB to a connected computer.

juani13973, 2021-06-10:
I bought forScore. It works great with the midi commands. Problem solved form me. The app is very powerful.

closed 2021-06-14 by alf45tar

T-vK, 2025-07-02:
@alf45tar Would you be interested in reopening this issue?
I saw you mentioned the ESP32-BLE-Keyboard library that I wrote years ago and that memory constrainsts were an issue. In 2022 somene added support for NimBLE reducing the memory usage. Maybe it would be worth reevaluating. 
Besides that, the newer variants like the ESP32-S3 have quite a bit more memory so maybe we could enable that featue based on the microcontroller used.