alf45tar/PedalinoMini #235: Keep leds on all the time

juani13973, 2021-06-10:
Hi alf45tar, I've been trying to play with the configs but I did not manage to keep a led of the footswitch on (independently of the state of the CC value).
In my 18 footswtich pedalinomini I am using several apps running in the iPad. I want to color code the footswtichs to associate them to an app. ex : Red is Bias FX-2, Blue is Quantiloop etc.
I want the led always on independently of the value sent. is this possible ?

with this config below, the led is on only when the footswtich is pressed, but I am saying the led should be on and the same color when off

[image]

With the config below. It toggles between on and off and the led is only ON when the value 127 is active. Said so, I said I want the led with the same color when off.

[image]

Conclusion, is it possible to define the color of the led and leave it always on independently of what I do with the footswitch configuration ?

alf45tar, 2021-06-11:
It should be possible. If not there is a bug.
I will check and let you know.

TarFilarek, 2021-06-11:
Hello,

I've checked it and it works on my controller. 

You should go to **Options** tab and check the **Leds Off Brightness**. Probably Your slider is on **0** position. 

[image]

You can use these sliders to distinguish ON and OFF states with the same colour LEDs (eq. light green when off, strong green when on). You can also put them in the same position, to make LEDs identical in both states.

alf45tar, 2021-06-11:
@TarFilarek very good insight

juani13973, 2021-06-12:
Thx TarFilarek and Alf45tar problem solved !!! All working now.
Quick one : is it my leds or the colors are inverted ? when I select Green, I get Red, and when I select Red I get Green.
That is easy but trying to guess the other 16M colors gets complicated :)
Thx again,

TarFilarek, 2021-06-12:
I'm glad it helped :)

I had similar problem with guitar effects in AmpliTube - they were turning ON on off button state, and vice versa. It caused LEDs to display inverted* colours. 

*They are actually not inverted. What's inverted is Your expected type of action triggered by button press. You expect it to turn something on, but it's actually turning it off.

Some digital actions are triggered with different physical button actions (press, release etc.), type of message sent (control change, program change etc.) and midi velocity values (0, 127). 

If Your colours are inverted, You can change it by:

- changing midi velocity values assigned to physical buttons in Pedalino (both boundry values to 127 or 127,0 instead of 0,127)

- changing type of action triggering physical buttons (eq. from press to release)

- changing type of midi message type (eq. from control change to program change)

- adding another physical button action for release state. It will "reset" midi velocity value after each press (it will send 0), and allow next press to send 127 (instead of sending 0, 127, 0, 127 repeatidly without release action, which would take care of sending 0 values)

- simply chaning Pedalino's OFF status LED colour to Your desired ON status colour and vice versa (so off state led becomes on state led etc.)

You should experiment with all of the above for the best results. Some tips will work great with loopers, other with actions like "Record" button or "Guitar Effects".

You can also download software like MidiMonitor (for Windows, but there should be something for iOS) to monitor midi values and observe caused reaction. It will help You understand how midi works with the software and modify Pedalino's setting for the best results.

alf45tar, 2021-06-14:
Replace

https://github.com/alf45tar/PedalinoMini/blob/c833f75fa3c3819ffb2cc6abd95c20f727c9b0eb/src/PedalinoMini.cpp#L302

with 

`FastLED.addLeds<NEOPIXEL, FASTLEDS_DATA_PIN, GRB>(fastleds, LEDS);`

closed 2021-07-29 by alf45tar