alf45tar/PedalinoMini #236: Configure the position of the highlighted effects

teopost, 2021-06-13:
In my pedalinomini, button 1 (bottom left) activates the chorus effect (top left of the display).
Is it possible to change the position of the effect Chorus shown on the display, on the botton left corner ?

[image]

alf45tar, 2021-06-14:
I could just switch the lines but I am looking for a more flexible solution.

TarFilarek, 2021-06-15:
A great option would be to have predefined text zones and the possibility to choose, what is displayed in each one of them.

Own zone name would be especially useful with Ladders, in which only the first button name is displayed. 

In the case of a blank name cell, the zone could use the name assigned to chosen Pedal. 

It would also be awesome, to let the user choose a number of displayed rows and zones in each row. 

The **BANK NAME** could also be modified, to occupy less space and allow to choose more rows. 

Examples:

 **3D Printed PedalinoMini Case** user could choose 2 rows with 4 zones in each row to display a real layout of buttons. 

[**juani13973**](https://github.com/juani13973/pedalinio-18-footswitches-prototype) could use 3 rows with 6 zones in each row, etc.

mycowx, 2024-03-14:
> Own zone name would be especially useful with Ladders, in which only the first button name is displayed.
> 

100% this.  Ladder mode only shows first button name.  Would be great if it could show all 6 (using the TC-Helicon Switch-6)